axoloti/axoloti #2: spat/pan m object doesn't compile.

vormplus, 2015-02-01:
Opened the first mini synth I made, and it doesn't work anymore. I tried recreating it step by step and found out the spat/pan m object is causing the error.

This is the console output:

	Generate code complete

	Start compiling patch
	RM
	rm -f xpatch.o xpatch.elf xpatch.bin xpatch.d xpatch.map xpatch.lst
	APP
	arm-none-eabi-g++ -nostdlib -fno-exceptions -fno-rtti -mcpu=cortex-m4 -O3 -fomit-frame-pointer -falign-functions=16 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Wunused-parameter -DCORTEX_USE_FPU=TRUE -DTHUMB_PRESENT -mno-thumb-interwork -DTHUMB_NO_INTERWORKING -mthumb -DTHUMB -std=c++11 -DARM_MATH_CM4 -D__FPU_PRESENT -I../CMSIS/Include -I../chibios/os/ports/common/ARMCMx/CMSIS/include -I../chibios/os/ports/common/ARMCMx -I../chibios/os/ports/GCC/ARMCMx -I../chibios/os/ports/GCC/ARMCMx/STM32F4xx -I../chibios/os/kernel/include -I../chibios/os/hal/include -I../chibios/os/hal/platforms/STM32F4xx -I../chibios/os/hal/platforms/STM32 -I../chibios/os/hal/platforms/STM32/GPIOv2 -I../chibios/os/hal/platforms/STM32/I2Cv1 -I../chibios/os/hal/platforms/STM32/OTGv1 -I../chibios/os/hal/platforms/STM32/RTCv2 -I../chibios/os/hal/platforms/STM32/SPIv1 -I../chibios/os/hal/platforms/STM32/TIMv1 -I../chibios/os/hal/platforms/STM32/USARTv1 -I../chibios/boards/ST_STM32F4_DISCOVERY -I../chibios/ext/fatfs/src -I../firmware -MD -MP -c xpatch.cpp -o xpatch.o
	xpatch.cpp: In member function 'void rootc::instancepanm__1::dsp(const int32_t*, int32_t, int32_t (&)[16], int32_t (&)[16])':
	xpatch.cpp:222:47: error: 'c' was not declared in this scope
	 outlet_left[buffer_index] = ___SMMUL((1<<29)-(c<<2),inlet_i1[buffer_index])<<2;outlet_right[buffer_index] = ___SMMUL((1<<29)+(inlet_c<<2),inlet_i1[buffer_index])<<2;
												   ^
	make: *** [all] Error 1
	shell task failed, exit value: 2
	Done compiling patch

vormplus, 2015-02-01:
And here's the XML from the patch.

	<patch-1.0>
	   <obj type="dac~1" sha="d26fce66143f96b96fa3df63ea01b45ff4b3bef7" name="dac~1_1" x="1240" y="20">
		  <params/>
		  <attribs/>
	   </obj>
	   <obj type="keyb" sha="b8deb97637e54be31fcb62e849e4fa406e72256e" name="keyb_1" x="20" y="60">
		  <params/>
		  <attribs/>
	   </obj>
	   <obj type="osc~" sha="57fd153c89df1299ed1ecbe27c961ac52732ab5" name="osc~_1" x="200" y="60">
		  <params>
			 <frac32.s.map name="pitch" value="0.0"/>
		  </params>
		  <attribs/>
	   </obj>
	   <obj type="*" sha="c47ceb7366785e0103cf880ce3450321491949f1" name="*_1" x="360" y="60">
		  <params/>
		  <attribs/>
	   </obj>
	   <obj type="echo1~" sha="7611c8729f6b16550b0764bc62bc93ccc71235fc" name="echo1~_1" x="480" y="60">
		  <params/>
		  <attribs>
			 <int attributeName="delaylength" value="4625"/>
		  </attribs>
	   </obj>
	   <obj type="*c" sha="d36ecbd55095f4888a0ebda8efda68e015c5e72b" name="*c_1" x="800" y="60">
		  <params>
			 <frac32.u.map name="amp" value="8.0"/>
		  </params>
		  <attribs/>
	   </obj>
	   <obj type="panm" sha="f2d9a02beeb3b451541dc262fc3d1e3b1d3af031" name="panm_1" x="960" y="60">
		  <params/>
		  <attribs/>
	   </obj>
	   <obj type="dac~2" sha="a23b68aa6010f51e12035361950edcd7afae239" name="dac~2_1" x="1240" y="80">
		  <params/>
		  <attribs/>
	   </obj>
	   <obj type="saw2" sha="44413381d696d82ab57e28ec39239114e5e61ba3" name="saw2_1" x="220" y="220">
		  <params>
			 <frac32.s.map name="pitch" value="24.0"/>
		  </params>
		  <attribs/>
	   </obj>
	   <obj type="osc" sha="6215955d70f249301aa4141e75bdbc58d2782ae6" name="osc_1" x="800" y="220">
		  <params>
			 <frac32.s.map name="pitch" value="-30.0"/>
		  </params>
		  <attribs/>
	   </obj>
	   <nets>
		  <net>
			 <source name="keyb_1 note"/>
			 <dest name="osc~_1 pitchm"/>
		  </net>
		  <net>
			 <source name="keyb_1 gate"/>
			 <dest name="*_1 b"/>
		  </net>
		  <net>
			 <source name="osc~_1 wave"/>
			 <dest name="*_1 a"/>
		  </net>
		  <net>
			 <source name="*_1 result"/>
			 <dest name="echo1~_1 in"/>
		  </net>
		  <net>
			 <source name="echo1~_1 out"/>
			 <dest name="*c_1 in"/>
		  </net>
		  <net>
			 <source name="saw2_1 wave"/>
			 <dest name="echo1~_1 feedback"/>
		  </net>
		  <net>
			 <source name="*c_1 out"/>
			 <dest name="panm_1 i1"/>
		  </net>
		  <net>
			 <source name="panm_1 left"/>
			 <dest name="dac~1_1 wave"/>
		  </net>
		  <net>
			 <source name="panm_1 right"/>
			 <dest name="dac~2_1 wave"/>
		  </net>
		  <net>
			 <source name="osc_1 wave"/>
			 <dest name="panm_1 c"/>
		  </net>
	   </nets>
	   <settings>
		  <subpatchmode>no</subpatchmode>
	   </settings>
	   <notes><![CDATA[]]></notes>
	</patch-1.0>

JohannesTaelman, 2015-02-01:
Fixed in commit dc443a67636bb4049e436390ffc67f270c8f0f83

closed 2015-02-01 by JohannesTaelman

vormplus, 2015-02-01:
Thanks for the quick fix!