axoloti/axoloti #3: Zooming in patch

JohannesTaelman, 2015-02-02:
Study feasibility - together with updating the object layout...

nicolasmiller, 2018-07-29:
There's an implementation of a zoomable UI in the experimental branch that I wrote. Under the file menu select open new Piccolo view. Use the mouse wheel to zoom. Mouse wheel click to pan. The Swing view is the standard unzoomable UI. The way it works currently is that you can have both view types open looking at the same patch and changes are reflected in both places.

simonvanderveldt, 2018-07-29:
@nicolasmiller Ah, cool. Sounds good! Are there any builds available for the experimental branch? Or would I have to build it myself? (I'm using Linux)

nicolasmiller, 2018-07-29:
I don't think we have an automated release build; we really should. We do use Travis CI to run a build/test cycle on each commit, but off the top of my head, I'm not sure if you can get anything useful out of there after the build has run: https://travis-ci.org/axoloti/axoloti/

The most immediate way is just to build from source which is pretty painless: https://github.com/axoloti/axoloti/blob/experimental/doc/compiling%20from%20source%20on%20linux.html

The issue is that experimental expects the hardware to be on that corresponding version of the firmware. It should detect this and prompt you to update. A related issue is that the libraries are bound to a particular patcher/firmware version. Basically it's currently painful to go back and forth between master and experimental for practical use. This is something that needs to be improved and has been in my TODO list for a long time. There is way too much complexity exposed to the user around library/patcher/firmware versions; it should just work. For the time being, if you're fine moving forward to experimental, that's great. But if you want to be able to move back to master, keep the libraries separate and be aware that you may have to fiddle with firmware to get everything working. 

Note: You can always rescue the board via DFU mode if something goes wrong. This is a method of putting the board into a special mode that allows the patcher to reset the firmware completely. You hold switch 1 during power on to get into it. Check the forums for more info or ping me if you need help.

Bottom line: I'd really like to get another high dpi user testing that stuff out and eventually get it into master. I wrote it initially because I was in the same frustrating situation with a high dpi screen. If you need me to make a build for you manually, let me know.

simonvanderveldt, 2018-07-29:
> I don't think we have an automated release build; we really should. We do use Travis CI to run a build/test cycle on each commit, but off the top of my head, I'm not sure if you can get anything useful out of there after the build has run: https://travis-ci.org/axoloti/axoloti/

That would be nice. Unfortunately Travis doesn't have it's own place to store build artifacts, so a build either needs to be attached to a release on GitHub which doesn't really make sense for development branches or another way to store the artifacts has to be be used like S3.

> The most immediate way is just to build from source which is pretty painless: https://github.com/axoloti/axoloti/blob/experimental/doc/compiling%20from%20source%20on%20linux.html
> ...
> Bottom line: I'd really like to get another high dpi user testing this stuff out and eventually get it into master. I wrote it initially because I was in the same frustrating situation with a high dpi screen. If you need me to make a build for you manually, let me know.

If you could create a build for the experimental branch similar to the release `.deb` that would be nice. Otherwise I'll spin up an Ubuntu container and see if I can get it to build there.
I don't mind to mess around a bit to try out the new version and give some feedback :)

nicolasmiller, 2018-07-30:
@JohannesTaelman Will check out the piccolo TODOs. It does at least load up and look to be generally working at the UI level on the latest experimental. I see one obvious visual glitch with a number being renderered near a toggle.

simonvanderveldt, 2018-08-01:
> HiDPI does not need scaling in the application, and works out-of-the-box on MacOS. I think Swing is failing to autodetect HiDPI on your linux. You could force java to do HiDPI as described here.
> Migration to JDK9 could be a solution, but unfortunately no JDK9 32bit builds are available...

I just tried this, but while using `uiScale` does result in a usable scale for the patcher area it also makes the rest of the UI rather oversized, so you get very large menu bars:
[image]
This is even more obvious on the main window, on which everything becomes too large
[image]
This window as well as the menu bars of the patcher look fine scaling wise (they have some other issues like incorrect height/wrong alignment) without the `uiScale` option
[image]

So the issue is mainly with the patcher area.

> @simonvanderveldt Feedback on the experimental branch is welcome. Currently, pre-build binaries from the experimental branch would probably be very short-lived, and need a build from git source to develop a fix for something, so I'd recommend git source builds.

OK, I'll see if I can build it myself :)

P.S. All screenshots are of full-screen windows

JohannesTaelman, 2018-08-01:
This issue seems to match https://bugs.openjdk.java.net/browse/JDK-8058742

simonvanderveldt, 2018-08-01:
> This issue seems to match https://bugs.openjdk.java.net/browse/JDK-8058742

Yeah, it sounds like that's the issue. Last update from 2016 though, so not holding my breath for a fix from them :P

> @simonvanderveldt Which desktop environment are you using? What desktop scale factor are you using? I'm a Linux user as well and normally use XFCE with 2x DPI on a high dpi screen. What's weird is that the patcher doesn't pick up on the GTK look and feel under XFCE. You get the AWT metal style. And it doesn't respect the desktop DPI scale factor. This is by default without providing any command line overrides like you were attempting to do. If I recall correctly, in GNOME, by default it will decide to use the GTK look and feel and also pick up the correct desktop scaling factor for fonts, but in the bug it sounds like they expect the opposite. Anyway it seems like the command line option is causing the fonts to be scaled twice while the patcher rendering is only scaled once. I'll see if I can reproduce and come up with a workaround on my machine.

@nicolasmiller I'm using GNOME 3.26, scale of 200%.
For XFCE it might be there's a check somewhere (maybe in the Java GUI libs?) that checks if it's GNOME instead of a DE that's using GTK?

simonvanderveldt, 2018-08-01:
If I just want to try the new zooming code in the experimental branch is it enough to replace `/opt/Axoloti/app/axoloti-app.jar` with `./dist/Axoloti.jar` that was created by running the build?
And then, looking at the settings, the scrolling should zoom in and out, correct?

nicolasmiller, 2018-08-01:
Need to double check how the actual release build is created. The easiest way is just to run ./Axoloti.sh after running ant from the experimental source directory. Running ant builds only the java sources. This assumes that you've already run build.sh on a fresh checkout.

Once you have the experimental client running, open an example patch normally, then from its File menu select new Piccolo view (Piccolo2d is the rendering tech used for the new canvas type). This will open a separate view of the patch that has mousewheel zooming/panning enabled. Changes should be reflected in both open views. You can close the original view and use the zoom view alone. It's just set up this way for debug right now. We might make some other way of enabling it in general later on.

simonvanderveldt, 2018-08-02:
@nicolasmiller Thx! got it to work :)
[image]

Regarding zooming: It seems to work/look fine :)
A setting for a certain zoom-level would be nice I think. Unless a fix for the HiDPI behavior would automatically fix the patcher view so it would be correctly sized automatically.

Some other observations:
- As you can see some nodes fall outside their object's box (see all objects in the second column)
- The font-rendering looks different. Don't know if that was planned or not. Especially noticeable with multiple underscores next to each other
- I don't get the pop-up to add new objects when double clicking in the piccolo view. Is that still missing functionality?
- Really like the additional options/windows in the "Board" menu, especially the MIDI monitor :+1: 
- Maybe the patch cords could be slightly thicker? They seem a bit tiny/1px wide

nicolasmiller, 2018-08-02:
Glad you got it working! These glitches are expected for right now; I'm
going to be patching them shortly. Johannes has been making some changes in
other places in the app that require some minor changes in the Piccolo
renderer. There are some minor visual differences because the entire
rendering pipeline is different. We can work on some of the usability
stuff. In general, the visual elements are a bit flatter in the Piccolo
view because some of the shadow stuff and embellishments on the standard
view look strange when zoomed. Like normally there's a shadow on the patch
cord originally if I recall correctly that I ended up removing.

We certainly could try to default to some zoom level depended on a detected
DPI. The safest is probably to get highdpi working in general and then let
the user zoom in and out from there.