axoloti/axoloti #9: uploading firmware, missing flasher.bin

kaosbeat, 2015-02-24:
currently unable to flash using SD, (fallback DFU seems to work)

```
Start uploading file /Users/kaos/Documents/axoloti/firmware/build/axoloti.bin to sdcard : firmware.bin
uploading: /Users/kaos/Documents/axoloti/firmware/build/axoloti.bin
uploading : 6%
uploading : 12%
uploading : 18%
uploading : 24%
uploading : 30%
uploading : 36%
uploading : 43%
uploading : 49%
uploading : 55%
uploading : 61%
uploading : 67%
uploading : 73%
uploading : 80%
uploading : 86%
uploading : 92%
uploading : 98%
uploading : 100%
Done uploading file
Start uploading patch
bin path: /Users/kaos/Documents/axoloti/firmware/flasher/build/flasher.bin
FileNotFoundException
```

JohannesTaelman, 2015-05-28:
"Make"-ing flasher sees objects in firmware/build and skips building them in flasher/firmware/build.
Does not create build directories.
And also broken because of changes in firmware.

JohannesTaelman, 2015-06-15:
Flasher now uses sdram (rather than sdcard) for temporary firmware storage. Build issue fixed.

closed 2015-06-15 by JohannesTaelman