axoloti/axoloti #11: stm32F4discovery won't respond when adding too much mod sources?

ramonschepers, 2015-04-07:
so, when i add more than 7 analog inputs (10k mono lin pots, all have different mod sources), and i went live on my board, the board hangs, and does not respond until i re-power it, the CPU usage is about 85% (with 7 analog inputs, running stable) so that's not a problem. update interval of each mod source is set to 103ms.
any ideas on this?

JohannesTaelman, 2015-04-07:
Is patch->settings_.maximum number of modulations sources set to >7? This limit is not checked or enforced.
Some pins on the discovery are not free, check with the discovery user guide.

ramonschepers, 2015-04-07:
oh whoops, i forgot to set the amount of modulation sources to be higher than 7, thanks for your help :)

closed 2015-04-07 by ramonschepers