axoloti/axoloti #14: GUI: new object dialog, resizeable/draggable

TheTechnobear, 2015-04-18:
make a floating window that can be dragged out the way, and resized

TheTechnobear, 2015-07-29:
I had a quick look at this, easiest way is to add the window decorations back, but this arguably makes the window a little bulky, and we'd also need to ensure it cannot be minimised.
an alternative approach, is to add a custom title bar (ie. thinner) or a drag box, this probably is best approach... not too difficult, but care needed. also we'd need to still reactivate the resizing border, or again, have a new drag bar for resizing.

toneburst, 2016-03-04:
I think the option to haver the object browser be persistent and draggable would be useful.
Quartz Composer, which I use to use a lot, and which has been mentioned several times in this forum does it like this:

[image]

The two small floating windows are the library (right) and inspector, where properties of the selected object are set.

a|x