axoloti/axoloti #35: GUI: circular references

JohannesTaelman, 2015-05-28:
crash when opening a patch with circular references to subpatches

TheTechnobear, 2015-07-10:
checked this out....
issue is due to the postConstructor resolving the type... which causes a new subpatch to be loaded again

Patch.PostConstructor calls ResolveType calls GetAxoObjectFromName calls AxoObjectFromPatch call PostConstructor

unfortunately the Patch cannot break the link, since it is not types, only has instances.

The best solution is that we only have one copy of the SubPatch type in memory.
this could probably be enforced in GetAxoObjectFromName, where we could cache on (exact) filename(?).
we would also appear to need to do this in AxoObjectUnloaded when we load it.
(again to ensure there really is only ever version of the type in memory)

This we have the great benefit, of only seeing the subpatch once in the search browser, and opens the door to fixing the issues related to editing the subpatch via different patch windows.

though rather than allowing many copies of patch open in GUI, id suggest if the patch is already open we merely bring it forward and giving it focus.... so we don't have to start doing redraws for other copies that may be open!

JohannesTaelman, 2015-07-10:
Ok for caching on filename with absolute path.

In the future, "live" could also be made to work on subpatches. But then when a certain subpatch-object is open as patch, it needs to be bound to the right parent patch, parameter-on-parent parameters disabled etc.

TheTechnobear, 2015-07-10:
K... we will need to check that we always have absolute name... 
might have to let the file system resolve it, so we get a proper path.  don't want ./test to be different to ../a/test from same patch - but don't see that as an issue.

subpatches, yeahI think we have a small inconsistency, but possibly desirable.

a)if you open explicitly, Live is not disabled BUT Ive now added a warning. 
I did this, as I think when your creating/testing a subpatch sometimes its useful to go live,
as we said before, because you temporary substitute the outlets with an audio out.
so I think warning is better than having to save temporarily as a patch.

b) if you open a subpatch via edit object instance, Live is disabled.
ok, I kind of agree, this would be nice to launch the parent patch
BUT its possible you could have 2 patches open, which both share the same subpatch,
ok, rare, but a use-case Id prefer to deal with, rather than ignore.

anyway I think ok for now.

will look at caching AXS types... though slightly concerned it MAY turn into a bigger change, and don't really want to break anything now, so close to release.

JohannesTaelman, 2015-07-26:
circular references are a bit academic at this point, closing this issue, saving #20

closed 2015-07-26 by JohannesTaelman