axoloti/axoloti #41: substitute turns object name into long "path/object" name

timvets, 2015-06-24:
to test this:
open example>demos>keyboard>poly arp demo>arpeggiator
now try to substitute the stringvoice object (I had made an adapted version, which is what I wanted to replace it with), or just with stringvoice itself...
(click the triangle to the left of the object name 'stringvoice' and select 'substitute')
this results in the stringvoice object (or its replacement) having a _very_ long name, including the whole path, and the object is stretched out to over 5 times its original width.
this is on Ubuntu 14.04.2 LTS (ubuntustudio)

TheTechnobear, 2015-06-24:
confirmed on OS X too.
it also occurs when creating a new subpatch, not just substitute.

TheTechnobear, 2015-06-24:
seems to be confusion where the path of the subpatch is compare to the containing patch, removed patch suffix and seems ok now, fixed, raising pull request now

commit 46e673ff (2015-06-24): FIX #41 : remove suffix from patch path
commit 548aa100 (2015-06-24): Merge pull request #42 from TheTechnobear/unstable

FIX #41 : remove suffix from patch path
closed 2015-06-27 by JohannesTaelman