axoloti/axoloti #45: copy of block of objects doesn't include connectors

timvets, 2015-06-27:
linux (ubuntu), git pull at 21:52 UTC+01:00 today:
when making a copy of a block of connected objects,
the objects in the newly created copy aren't connected.
(some connections are there, others are not)

JohannesTaelman, 2015-06-27:
ack, happens when there's a space character in the object instance name

timvets, 2015-06-27:
you mean underscore? (there weren't any whitespaces in instance names)

TheTechnobear, 2015-06-27:
simple test case

math/+1 to gain/vca (v)  wont restore connections

clipboard looks ok, so i guess its the restore...
seems to be some objects... e.g. vca I cant get it to restore any connections.
(I cant find a pattern of which objects have issues though, and which don't)

<patch>
   <obj type="math/+1" sha="61413e41a22dba4d3f6a4be145c06146934f7120" name="+1_1" x="210" y="336">
      <params/>
      <attribs/>
   </obj>
   <obj type="gain/vca" sha="6bbeaeb94e74091879965461ad0cb043f2e7f6cf" name="vca_1" x="322" y="336">
      <params/>
      <attribs/>
   </obj>
   <nets>
      <net>
         <source name="+1_1 result"/>
         <dest name="vca_1 v"/>
      </net>
   </nets>
   <notes><![CDATA[]]></notes>
</patch>

TheTechnobear, 2015-06-27:
got it... its with inlets/outlets with 1 char names.. will check in a fix now

commit 454a1173 (2015-06-27): Merge pull request #47 from TheTechnobear/unstable

GUI: fix #45 pasting patches with inlets/outlets breaks connection when na…
commit d28ebbcc (2015-06-27): fix #45 pasting patches with inlets/outlets breaks connection when name = 1
closed 2015-06-28 by JohannesTaelman