axoloti/axoloti #48: midi/in/cc default value not sent

timvets, 2015-06-27:
When you set a default value in the midi/in/cc object, 
it is not sent to its outlet when going into live mode.
(or is this by design?)

TheTechnobear, 2015-06-27:
looking at object, it should send default when created, and it doesnt
(to test, I simply hooked a in/cc to a disp p to see if it would change the value and it didn't)
Johannes, the init code is ccv=%default% where default is an attribute is this still correct?
(I'm thinking possible regression due to the removal of %name%) 

tim... do you know if this works on the master branch?

timvets, 2015-06-27:
no, it doesn't work in the master branch either.

TheTechnobear, 2015-06-27:
k, thanks... 
one thing I noticed on my test, was it appears to send zero when patch goes live.
(and this does not change even if you change the default of the attribute to something else)

also midi/in/cc i works as expected.
whats strange is that the generated code for cc and cc i looks the same ?! 
Im guessing some type conversion issue?

TheTechnobear, 2015-06-27:
ok, got it... data type issue as suspected...

commit 34ca310b (2015-06-27): fix #48 - midi/in/cc not sending default
closed 2015-06-27 by JohannesTaelman