axoloti/axoloti #51: Axoloti main window jumps on top when setting patch live

timvets, 2015-06-27:
In Linux, when going into live mode, the Axoloti main window jumps on top.
This does not happen when running as "sudo ./Axoloti.sh" (which was the only way before adapting 49-axoloti.rules)

JohannesTaelman, 2015-06-27:
Is bug recently introduced?
Main window jumping to top is supposed to be triggered by severe errors (in red) - like a compile fail. That's where you'll find the compiler error anyway.

timvets, 2015-06-28:
indeed there is some red in the main window:

null
Generate code complete

are the first two lines after going into live mode. 'null' in red
It nevertheless successfully runs the patch.

timvets, 2015-06-28:
no 'null' when running as root.

JohannesTaelman, 2015-06-28:
So it's a matter of finding the origin of the "null" error. Do you get an exception in the terminal (from where you launched Axoloti.sh)?

timvets, 2015-06-28:
hmm, maybe still some permission problem:
SEVERE: null
java.io.FileNotFoundException: patch/xpatch.cpp (Permission denied)

JohannesTaelman, 2015-06-28:
You need to transfer file ownership of the whole tree to your user. Running axoloti as superuser created files your user does not have permissions for.

timvets, 2015-06-28:
ah, the reason was because I've been running Axoloti.sh as root before.
removed patch/xpatch.cpp, which was owned by root, and now it can create a new xpatch.cpp owned by user.

closed 2015-06-28 by JohannesTaelman