axoloti/axoloti #52: Axoloti asks: Do you want to save changes... , always

timvets, 2015-06-28:
Even when nothing has been changed to the patch, Axoloti asks "Do you want to save changes...".
(linux)

TheTechnobear, 2015-06-28:
just tested
if you open and close patch, you do NOT get it.
BUT if you take it live you do... so its something to do with locking/live patch... just going to check what it is.

TheTechnobear, 2015-06-28:
ok the issue is parameter changes cause the patch to be marked as 'dirty'
when you take the patch live this happens twice:
a) showPresets... so when the patch is taken live, it defaults all the parameters to the preset value, which marks it as dirty
b) the axoloti board also appears to be setting the parameters on startup, as we get a parameter change for each.

SO, @JohannesTaelman, I think the underlying issue is:
Generally parameter changes should NOT mark a patch as dirty... 
the only reason we want this behaviour, is since when you are editing a patch, you are setting the parameters to give them a default.

Therefore, I propose setDirty should only ever change dirty state IF the patch is in edit mode aka Locked.

this means if its Live no changes will be saved which I think is correct... since the board may not even be connected to a computer at the time, i.e. LIVE = disconnected from patch editing.

TheTechnobear, 2015-06-28:
ok, I tried the above...
Generally it works well  BUT highlights another issue....

if you change the value whilst you are LIVE, then when you come out of LIVE the parameter cannot be reverted to the value that is 'saved in the patch' i.e. what I was referring to as the default/startup value.

this means, that the value whilst unlocked is not the last LIVE value, and you will not be asked to save, which is inconsistent.

as an alternative, I tried to clear the dirty flag in 'GoLive' / QCmdLock() but that doesn't work as (b) above is asynchronous.... and you don't know now, if the param changes is due to startup or is later down the line.

my opinion is... the 'fix' is correct, parameter changes done whilst live should not be considered to be changing the patch. and that we should implement this. 
but then we should raise another issue, which is parameters should be restored to saved values when patch is unlocked.

thoughts?
@JohannesTaelman if you agree with this, then I can checkin the SetDirty/ClearDirty changes.

JohannesTaelman, 2015-06-28:
hmm things can get confusing
The question is:
When is tweaking a patch live considered playing the patch or improving the patch?
Both need a different behavior, and I believe only the user knows the answer. The answer may be different for different adjustments. 
The quick and safe solution is setting the dirty flag - keeping it as it is. Unintentionally losing changes can be frustrating. 
Maybe a popup dialog after exiting live asking to keep the changes or revert them, maybe with a full list of adjustments? But I thing that quickly gets annoying. And deep work to implement.

timvets, 2015-06-28:
If you run the patch without tweaking anything, it should not ask if you want to save changes.
But I see your point.
It would be a good thing, if we're going to get this window each time, that you can do 'TAB TAB ENTER' to close without saving, without having to point-and-click each time. (right now those keystrokes have the same result as clicking 'cancel')

TheTechnobear, 2015-06-28:
an idea... perhaps I can change the param code to only setDirty if the value has actually changed?
(might be a slight issue with floats, due to accuracy but other types should be ok)

JohannesTaelman, 2015-06-28:
@TheTechnobear that's a good solution

TheTechnobear, 2015-06-28:
done and resolved, pushing now

commit 886cd9c8 (2015-06-28): Merge pull request #55 from TheTechnobear/unstable

fixes #52 & closes #52 only setdirty if parameter value changes
commit bb9b7091 (2015-06-28): fixes #52 & closes #52 only setdirty if parameter value changes
closed 2015-06-28 by JohannesTaelman