axoloti/axoloti #53: default window size of patches?

timvets, 2015-06-28:
When opening e.g. a demo patch for the first time, the window size is quite small.
You have to resize it to see the whole patch, then save it, to get it to open the right size next time.

TheTechnobear, 2015-06-28:
| think this is just a 'transitional change',  we just need to go open all demo patches resize and then save... I don't think its worth a code change for something that will later no longer be used.

I can do this, if agreed...

Note: I have seen that if the size IS present, then the master branch can no longer open the patch.
(so a patch saved on unstable, can no longer be opened in master  ... vice versa is ok)
but again, i think just a transitional issue... assuming unstable is going to be merged to master for release.

JohannesTaelman, 2015-06-28:
I tried auto-sizing the patch window when no size is saved but this is not working. The relevant code is in the PatchFrame constructor.

Unstable is going to be merged to master for release. I would not want to miss all the recent improvements.

TheTechnobear, 2015-06-28:
yeah, I hate java windows handling :) 
the issue is, its not calculating the viewport size correct in adjustSize....
I guess somewhere you must be trying to determine the best size, and setting as a preferred size for the canvas? 
(size should be presumably  lowest postion of an object and furtherest right of other objects, plus of course the objects height/width as appropriate)
after that, I think you should be able to call pack() to let it determine the size.

the issue with java, is if you don't get all the preferred sizes/layouts correct ... it ends up a mess :(

JohannesTaelman, 2015-06-28:
I believe adjustSize works fine as it is responsible to adjust the scrollbar range after moving an object.

yes, size/layouts handling is messy

I'm on this one now

TheTechnobear, 2015-06-28:
Hmm I looked at the return dimensions after ..and its square (incorrectly)
... perhaps at this stage, the objects have not been layed out?

TheTechnobear, 2015-06-29:
k, fixed it.. checking in

commit 003192a3 (2015-06-29): fixes #53, calculate initial size based on objects locations
commit d77c625a (2015-06-29): Merge pull request #57 from TheTechnobear/unstable

GUI: fixes #53, calculate initial size based on objects locations
closed 2015-06-29 by JohannesTaelman