axoloti/axoloti #54: Board selection

JohannesTaelman, 2015-06-28:
Replacing the serial port selection dialog...

TheTechnobear, 2015-06-28:
is this going to allow multiple board connection via USB connections .... or not quite yet :)

JohannesTaelman, 2015-06-28:
different options for multiple board connections.
libusb could tell which port it is attached to
or the cpuid could be exposed in the usb descriptor
2nd would be preferred but that makes the usb descriptor a dynamic string
either way is going to be a bit cryptic

Not very high priority. Removing visible references to serial port is more urgent.

TheTechnobear, 2015-06-28:
Yeah 2nd is preferable if you want different boards for specific purposes (
quite likely)
What about defaulting to cpuid but allowing it to be renamed and saved in
firmware?
( hmm, seems similar to board settings we were discussing re: input gain
etc)

Btw it might good idea if your refactoring to consider board selection will
at some point be required at patch window level ... So you can have
multiple boards connected and running patches.
( again not for now feature, but perhaps worth hating in mind )

Don't think many of us order multiple boards :) but hopefully that could
change once everyone is blown overboard by axoloti

JohannesTaelman, 2015-06-28:
ack

commit 6b9a25ae (2015-07-13): GUI: board selection via cpuid, fixes #54
closed 2015-07-13 by JohannesTaelman