axoloti/axoloti #56: request: keystrokes for "Don't save"

timvets, 2015-06-29:
The dialog "Axoloti asks: Do you want to save changes to..." can now only be closed by clicking.
It would be nice if the keystrokes "tab tab enter" would do the same as clicking on "Don't save" and "tab enter" == clicking on "Save".

JohannesTaelman, 2015-06-29:
This works in Windows. If it does not on linux, the issue is in java/linux.

TheTechnobear, 2015-06-29:
perhaps a linux issue... 
n mac OS X - tab correctly cycles thru button, and space (correctly) activates the chosen option.
enter selects the default option (cancel)
and escape selects cancel.
this all confirms to UI standards.

Id say the only thing that is 'odd' is ENTER should be SAVE, i.e. save is the default.
(agreed?) 

@timvets can you test on linux that tab is not working at all? perhaps window manager setup?

timvets, 2015-06-29:
Yes tab is working, and ...:
It _does_ work with SPACE !
usually ENTER is what I use..., and I still think ENTER would be more intuitive.
but I can live with SPACE :)

TheTechnobear, 2015-06-29:
nah, UI standards dictate ENTER = default option.  
hence the 'debate' about what the default option should be :) 
I checked with other apps, and does appear save=enter, escape = cancel  
... and do not save, is tab, we could change the order so that cancel is next to save, so its "do not save" is tab then space.

JohannesTaelman, 2015-06-29:
Enter/space works on windows too. 
I believe this issue is so tiny that I close it.

closed 2015-06-29 by JohannesTaelman