axoloti/axoloti #62: GUI : warn when opening patches intended for subpatch use

JohannesTaelman, 2015-07-04:
Warn the user when opening a patch containing inlet/outlet objects, not to expect sound...

TheTechnobear, 2015-07-04:
hmm, i think this would get tiresome....

an idea, perhaps it should be a warning if you take a patch LIVE that has no audio or midi output objects?    (or just outlets) 
... as there is not really any reason to take such a patch live.

JohannesTaelman, 2015-07-04:
No audio or midi I/O is not indicative, inlet/outlet objects are. There is so much more that can be done in patch. Currently patches with inlets/outlets do compile and run, only with inlets and outlets ignored. It can be useful to diagnose problems in a subpatch.
I agree issuing the warning when going live is probably the best. After loading, the patch can still be modified, so load-time is wrong. 
A dialog or a red "severe" log message that pops up the mainframe?

TheTechnobear, 2015-07-04:
perhaps the warning should be more like....
"patch has unconnected inlets/outlets, not intended as runnable patch?"

I think should be in log.
(I don't think we should be popping up dialogs generally) 

longer term, I think that the log should be a console attached to the patch window... a bit like the console in an IDE... this will be much more logical if you have different patches running on different boards.

TheTechnobear, 2015-07-04:
"I think that the log should be a console attached to the patch window." ... if you agree with this, I could take a look at doing this... 
would be a necessary step to multi-boards and I think generally people will prefer it . (sorry I don't like the current floating console window :) )

JohannesTaelman, 2015-07-04:
Attaching the log to the patch window: I agree, but it requires a lot and very deep changes. I propose to keep that for the long term.

TheTechnobear, 2015-07-12:
this can be closed.

commit 5a789a4f (2015-07-10): GUI: warn if trying to take subpatch live fixes/closes #62
commit 8972935f (2015-07-10): Merge pull request #89 from TheTechnobear/unstable

GUI: warn if trying to take subpatch live fixes/closes #62
closed 2015-07-12 by JohannesTaelman