axoloti/axoloti #64: show subpatches from search path in object selector

JohannesTaelman, 2015-07-04:
like chorus2, flanger, compressor1

preloading all patches in the search path would be too heavy, only adding the object id in the object selector. May need a new derivative class of AxoObjectAbstract.

TheTechnobear, 2015-07-04:
this would be useful...
(we may need to organise the subpatches directly to be a bit more hierarchical )

the same thing could also list patches in the current patches subdirectory... this would be handy for selecting voices you have already saved.

so your saying a AxoObjectPatchReference (or similar) which only gets resolved when its actually required (=Loaded)

JohannesTaelman, 2015-07-04:
AxoObjectUnloaded is just comitted
first examples are in the fx/ category
the patches/subpatch folder is a mess, with a few useful things in between...

TheTechnobear, 2015-07-05:
oops forgot you moved,  chorus.axp	chorus2.axp	flanger.axp
can you change these all to .axs

I guess, once we have moved anything useful out of subpatch, we can then also move the remaining into archive, and remove patches/subpatch from the search path?

so the patches directory, is then solely used for demos and tutorials.

could idioms be move to tutorials? or move it, and patches/unsorted into the archive for the moment?

(I think the value at the moment, is doing the remaining tidying up of subpatch/demo/help ... and perhaps adding a few help files where they are really needed)

JohannesTaelman, 2015-07-05:
patches/subpatch will just get empty
idioms can move to archive

OK for moving it to the help menu. 

The main issue with help is that the root index is very cryptic, showing the html files in the categories somehow could help.

closed 2015-07-05 by JohannesTaelman