axoloti/axoloti #69: converting patches in patches/test to .axh help patches

JohannesTaelman, 2015-07-04:
The object generator code may check for presence of an equally named .axh patch, and
if not present, it could check for presence of an .axh patch named to the category. That patch could be linked as help in the .axo.

Or would it be better to do it at object load-time rather than at object generation time?

TheTechnobear, 2015-07-04:
hmm, the test files do look like good source for help files.

a few things:
- id be tempted to rename the test files to 'help files' ... this will make users far more likely to look at them.
id never bothered, because I assume they were not really interesting to an end user... just the writer of the object... and I can see I was wrong!

- I dont hugely like the axh file extension... because at the moment its identical to normal patch.
(I wonder if an axh will ever differ from axp?)

I wondering therefore if perhaps it make sense to:
- rename the patches/tests to patches/help
- move help files in to this directory
- in an AXO when we see helpPatch look in patches/help as a search path.

I think then we can easily just update the axo (via the generators) to take point to a relevant help file.

the other thing we might want to consider is an approach similar to Max,  with its packages etc.

what we do so is say that the things listed in the search path are 'package' directories.
a package directory then can contain the following directories 
- objects (for axo)
- help (for help patches) 
- examples (for example patches) 
- resources (for things like samples)

so the stuff supplied with axoloti is the 'core package'

but other users can then distribute their own packages, which can be installed where ever they like, 
and also can include objects,help and examples.
(the key thing is not mixing with the core package, so they can easily be upgraded etc, and also not 'destroyed' if axoloti is updated etc)

we are not very far from this already (just need to move a few things around, and make a few menus 'dynamic' ) , and I think it could be quite a powerful concept...

JohannesTaelman, 2015-07-04:
I think the objects folder should be able to contain a mix of
* coded objects (.axo)
* patches for use as subpatch like a regular patch (.axp)
* respective help patches (.axh)
If help patches would be marked with a field in the XML, their XML would need to be parsed before knowing, could get sluggish..

I can't think of a need to use a different file format for a help patch.

Basically it is distinguishing patches and help patches by extension rather than putting them in separate directories. I seem to prefer having the help files sit in the same folder as the objects.

Yes things are converging quickly!

TheTechnobear, 2015-07-04:
yeah I prefer code and help/examples to be in separate places... but the follow the same hierarchy
its a pretty common 'idiom' in coding... we do it with source files vs object files, and usually help as well.

the thing is 3 axo may all use the same help file... so that quickly breaks the name.axo = name.axh, which means you start having a whole bunch of .axh files which are in the same directory as the axo but not obviously linked.
It also means you cannot just 'browse' the help files, by looking in one directory. (as it now littered with .axo)

( actually on reflection its not the axh extension that I don't like, it probably more is the fact its in with my source code :) ) 

one other thing that strikes me....
I would actually like subpatches to be seen more like axo  than as patches!! (from a user perspective), this relates to issue #64 
basically they are 'macros' (as reaktor would call them) , but when you use (as oppose to create them) them you don't care if its an axo or a subpatch - thats an 'implementation detail'.
(do I care if a chorus object is an axo, or a macro? no... I just want a chorus... it just happens that 'higher level' objects tend to be macros)

so I wonder if we should have a different extension? (axm ?) 
this might also address you other concern, about warning about taking live...

JohannesTaelman, 2015-07-04:
name.axo does not need to correspond with name.axh

name.axo has a String field "helpPatch" with a relative path to the .axh
so multiple objects can refer to the same help patch

I'd suggest adding in gentools.PostProcessObject() 
If the helpPatch field is still empty, and there is a equally named .axh present, to fill in the .axh field
if there is no name.axh, it could also test presence of category.axh

What's the advantage of parallel hierarchies? It's inconvenient in a file browser to navigate from a/x/y/z to b/x/y/z while it's easy to filter by extension. Separating from source code? Well the .axh itself is not generated output. OK .axo's are generated output but they're also in the repo, I do consider them documents, with a tool to generate them.

Check the presence of fx/chorus2 in the object browser. 

Towards macro support, it needs "propagate-display-to-parent", "propagate-attribute-to-parent", and then something like a front panel design tool... 
The front panel designer can converge with a object frontpanel design tool.

JohannesTaelman, 2015-07-05:
Differentiating between a macro and a patch, why?
A patch may have a "frontpanel" view without being a macro. No difference.
I could think of making a

TheTechnobear, 2015-07-05:
because they are conceptual different to the user
a macro cannot be used on its own, it cannot be taken live.

this is about 'user experience' (UX) sometimes, it doesn't matter if
implementation is the same, its how a user sees it...

in practice I tend to find that things that have conceptual difference,
whilst in early days may be similar (and share implementations), over time
tend to diverge.

they are already kind of different, if we look at the way patch.java has to
handle them, similarly the patch settings are sometimes going to only make
sense as a sub-patch.

also going back to the UX, when a user is creating patches, they want to
include sub-patches but never a main-patch.
I find it tricky at the moment going thru the demos, knowing what is a main
patch, and which is a voice... especially if they are not named well.
if we have a separate 'type' then ...
- the demos only have to show .axp files
- the open dialog can have a drop down to filter by .axp or .axm (yes we
should still be able to open axm)

(p.s. I plucked 'macro' out of Reaktor, perhaps a different name would be
more appropriate .. module? subpatch (axs) )

Mark

JohannesTaelman, 2015-07-05:
So I propose .axs for Axoloti subpatches

TheTechnobear, 2015-07-05:
underway...

commit 206a1132 (2015-07-05): rename examples menu to library, and populate with help files, fixes #69
commit ba40155a (2015-07-05): Merge pull request #74 from TheTechnobear/unstable

move tests to help files and link to relevant axo - fixes #69
closed 2015-07-05 by JohannesTaelman