axoloti/axoloti #84: OBJ:keyb mod - modulators broken

TheTechnobear, 2015-07-09:
caused by attrib change %name% to attrib_name

error:
xpatch.cpp:131:59: error: 'MODULATOR_attr_name_releasevelocity' was not declared in this scope
   PExModulationSourceChange(&parent->PExModulationSources[MODULATOR_attr_name_releasevelocity][0],NMODULATIONTARGETS,_rvelo<<20);

incorrect code:
`
src//main/java/generatedobjects/Midi.java:                + "  PExModulationSourceChange(&parent->PExModulationSources[MODULATOR_attr_name_velocity][0],NMODULATIONTARGETS,_velo<<20);\n"
src//main/java/generatedobjects/Midi.java:                + "  PExModulationSourceChange(&parent->PExModulationSources[MODULATOR_attr_name_note][0],NMODULATIONTARGETS,_note<<20);\n"
src//main/java/generatedobjects/Midi.java:                + "  PExModulationSourceChange(&parent->PExModulationSources[MODULATOR_attr_name_releasevelocity][0],NMODULATIONTARGETS,_rvelo<<20);\n"
`

should be getting the objects instance name... Im just not sure what this is in the generator, otherwise id fix it.

PLEASE NOTE: could also be a problem in:

`
src//main/java/generatedobjects/Patch.java:                + "            PExModulationSourceChange(&PExModulationSources[MODULATOR_attr_name],data2<<20);\n"
src//main/java/generatedobjects/Patch.java:        o.sKRateCode = "if ((%trig%>0) && !ntrig) {PExModulationSourceChange(&parent->PExModulationSources[MODULATOR_attr_name][0],NMODULATIONTARGETS,%v%);  ntrig=1;}\n
`

JohannesTaelman, 2015-07-09:
looking at it

JohannesTaelman, 2015-07-10:
There is still a problem with the modulation sources:
Adjusting one modulation target amount, sometimes reflect on the wrong source.
Also pasting a modulation source objects does not add its modulation source(s).
Deleting a modulation source objects does not remove its modulation source(s).
This is not related to the %attrib% changeover.
While I believe the modulation mechanism itself is fine: library/patch/modulator shows good stuff is happening.

TheTechnobear, 2015-07-10:
Cool, we can raise separate bugs for these issues, and sort them out.

Now you mention cut n paste , I found one annoying bug
Cut n paste a single connected object doesn't work. Think a null is thrown
too
( I use this to duplicate objects ) - only works if object is disconnected.

Also, do u mind if i make it so that pressing enter brings up new object
too, reaktor has this and it makes it really quick to enter new objects
from keyboard.
( max uses N with no modifier )

commit 31270398 (2015-07-12): GUI: fix #84
closed 2015-07-12 by JohannesTaelman