axoloti/axoloti #85: OBJ:env/line 2 x   - param fix

TheTechnobear, 2015-07-09:
hi

Ive noticed this error in other objects in the initcode we have 
`
phase1 = 0;
val = %v2%;
ntrig = 0;
`

the issue is the init code is not substituted, 
now I could change this so it is, so we get

`
phase1 = 0;
val = param_v2;
ntrig = 0;
`

but this will not work, since param_v2 is not in scope

so the question is, what do we set val to, obviously in this case I could set it to zero..
but that may not be the default of the param.

but does this matter, is the init value relevant, or will the code alway set val from param_v2 when its needed.
I slightly concerned that changing these initcodes() code introduce other subtle bugs.

possibly the other way to answer this is... before param_v2 we had %v2% , what would this value have been during the initCode()?

JohannesTaelman, 2015-07-09:
Parameter value was accessible in init code before, even though it's value may not have been initialized.
I prefer the stricter policy like it currently is, no access to parameter values during init.
Set it to zero, or leave it uninitialized.

TheTechnobear, 2015-07-09:
Cool, that's what I've done previously - just wanted to check there would
be no new bugs :)
Will fix in a short while.

commit 4d171a0f (2015-07-09): fix env object init code fixes #85
commit 6eb005c0 (2015-07-09): Merge pull request #86 from TheTechnobear/unstable

fix env object init code fixes #85
closed 2015-07-09 by JohannesTaelman