axoloti/axoloti #100: OBJ: Smooth broken

JohannesTaelman, 2015-07-12:
awkward scaling

TheTechnobear, 2015-07-18:
I guess distortion is done (saw checkin) 
will do the moving/removing ... 
im also going to rename flashread2~~ to flashread

also I assume, we really should rename anything which appears to have 'version numbers' like your play examples above, and X versions...  (x versions should also have x removed)

at same time I'll go thru helps/demos/tutorials to check these are using correct forms where appropriate.

JohannesTaelman, 2015-07-18:
Yes, great!
Yes, version numbers should be removed.
tilde suffixes too
"x" versions are the right versions

TheTechnobear, 2015-07-18:
ok, are you not keep tilde to differentiate audio rate objects ?(as max does)

does flashwaveread replace flashplay

TheTechnobear, 2015-07-18:
briefly, what do you mean by 'awkward scaling'  (Ive not looked yet, so may or may not be obvious :) )

JohannesTaelman, 2015-07-18:
All occurrences of ParameterFrac32UMapKLineTimeReverse have awkward 1/x scaling: infinity at 0...

Tilde suffix is perhaps familiar to max/pd users, but I believe overloaded objects are much better. As long as the differentiation can be observed from inlet types. Differentiating between oscillators and LFOs need to happen through naming, because the only different type is the outlet, I think osc/* and lfo/* are clear.

Tilde is also a very annoying character to type on different regional keyboards, often it is a dead character. 

wave/flashread2~~ -> wave/flashread interp
would be most consistent. 
But flash reading/playing are on the way out anyway. For simple sample playback, samples should be loaded from sdcard into sdram, and played from there.

TheTechnobear, 2015-07-18:
table/play 3 and table/record 2 ... replace previous versions (1,2 etc)

TheTechnobear, 2015-07-18:
done
note quantizer -> move to harmony category
timer/delayedpulse : remove non-x variants
timer/pulselength : remove non-x variants
wave/longdelay : remove
wave/play remove
wave/play2 *: remove
stomps/distortion1 broken  (done?)

to do 
pulse/d *, pulse/ahd *, pulse/hd *: awkward scaling
smooth: awkward scaling
change/speedlim : awkward scaling
remove table/play, play 2 and record  , replace with play 3/record 2? 

other 'oddities'... I'm not sure about
- tillde usage
vcf~ , I guess we will need to transition to ensure fewer breaks.
sampleshold~
xfade~
lfsr~
- fx/chorus, fx/chorus 2 & stomps/chorus 3  (they are all different, but number is not a good way to distinguish)
- stomps - remove 1 suffix?

TheTechnobear, 2015-07-20:
tilde object all gone, demos etc updates (tillde usage vcf~ ,  sampleshold~ xfade~ lfsr~)

closed 2015-07-20 by JohannesTaelman