axoloti/axoloti #93: Obj: Parameters to all have inlets?

TheTechnobear, 2015-07-12:
This is just an idea... for 'discussion' (may be relates to issue #12 )

I personally prefer objects to have inlets, so that I can programatically control, either algorithmically or from a UI comment (which may vary, e.g. a dial or a sequencer) - but its pretty common to then just attach a dial to that inlet to control from the UI.

the issue is, we currently have quite a few objects, that have parameters that are not also exposed as inlets, and this can be frustrating for a user (it has been for me a few times :) ), and also can lead to more object variations being created one with a parameter the other with an inlet.

the idea is that whenever a parameter is created on an object, and inlet is also created for that parameter. rather than the axo have to do this manually.

of course the question is what to do when you have an inlet connected, do you 
a) override the param, which becomes 'display only'
b) use the inlet as sum/offset on parameter

how to implement:
- the summing/override could possibly be an attribute of param, as it may be useful to be different for different objects.
- does the axo code, just see param_name, even for the inlet, or param_name and inlet_name... id prefer the former, so object does not have to specifically code...

backwards compatibility:
ok, we have to be careful here, as the additional inlet could already exist... 
options are:
- make a flag (that defaults to false) that allows parameter to not create inlet, so we can use where we see this use case.
- flag as error in generator, so we can fix.

thoughts?

notes:
- I know we have modulation sources, but I prefer connections, as I can see them easier when looking at patches
- only for parameters, as attributes are essentially 'compile time' constants'

JohannesTaelman, 2015-07-12:
I agree such a mechanism could clear up the object library quite a bit, however it is a significant change. Here's an overview of the issues involved.

## pfuctions ##

or "parameter functions" are pure functions, supplied as a function pointer in the ParameterExchange_t structure, that rescale the parameter value to the right dsp coefficient. The pfunction is called only when a parameter is effectively changed. This reduces the dsp load. 
This is why modulation sources require a trigger.

## When a parameter does not have a pfunction associated

* Exposing an unconnected inlet is not free in terms of cpu load. After adding inlet+parameter it needs to (signed or unsigned) saturate the sum to guarantee no overflows are happening. Even if it is only a single instruction, its an extra instruction cycle for every parameter involved in the dsp loop.
* Actually no pfunction associated rarely happens, since it defaults to a (unsigned or signed) saturation function ("pfun_signed_clamp" and "pfun_unsigned_clamp"), it is needed because parameter modulation can cause out-of-range values.
* It does cost screen space, some objects would become considerably more bulky. Packing a maximum of relevant information into a minimum of pixels

## When a parameter has a pfunction associated

* the scaling of the inlet would be weird if it is added after the pfunction. In this case, the pfunction should be disabled (or only do saturation), and the same (mathematical) function (different interface) should be applied within the dsp thread (preferably inlined).

## Separating parameters from objects?

Parameters on an object allow to specify unit conversions. When detached from an object this could
* explode the ctrl/dial objects to be specific for lfo pitch, osc pitch, amplitude, time and the user could connect the wrong one, causing a mismatch.
* this could be avoided by back-propagating the conversion from object inlet to "parameter object" outlet. But what to do when a single "parameter object" outlet connects to two destinations with different conversions?

## Summing or override inlet?

Current objects that have a modulation input do summing. This allows the user to check the expected influence in real world values by manipulating the associated parameter dial, and tweak the offset. 
With an override, yes, the parameter dial could turn into a display with unit conversion. 

## Approach

I'd like to preserve a case where a parameter can be used as in dsp as a coefficient with only a memory load, no other operations. I think the right solution is to introduce parameter modes, allow the user to choose between a regular parameter, parameter+modulation inlet and override inlet. It may be an implicit mode switch, that does the "parameter" only situation when the inlet is unconnected, but this situation needs to be handled in code-generation. The association between inlet and parameter needs to be visible in the object layout, current object layouts do not allow this.

While I'd like it to happen, this is not an easy change. I think it will need a branch for some time.

JohannesTaelman, 2015-07-12:
relates to #12