axoloti/axoloti #94: create native application and installers

TheTechnobear, 2015-07-12:


TheTechnobear, 2015-07-12:
ok, checked in first round of this... which uses ant:javafx
currently it builds bundle, only when asked explicitly too (just to keep compile time down, and in case of failure!)

use:
ant -Dbuild.bundle=true

it wont run currently, because the current working directory is in the application directory.
on a Mac this is: (it'll be similar on windows) 
Axoloti.app/Contents/Java/

Ive tested, if you copy the relevant axoloti files into that directory (which is not the desired solution!!!) it will work.

Can some one test this works on windows and linux ... as an ant/java task it is supposed to!!!

TheTechnobear, 2015-07-12:
next tasks are to look into moving things around a bit....

anything that is 'read only' could be added to the resources, and I believe that will make it work (i'll try) 

anything that is 'writeable' e.g. patch directory, should really be in the users 'document' folder.
obviously we will also need to be careful with compile scripts if directories are 'moved' around

TheTechnobear, 2015-07-13:
ok, stage 2... part 1 .. NOTE: we are in a 'transition' see below

latest checkin means that Axoloti users java (and system) env vars for locations.
(these are all defaulted to current location, but can be overriden)

these are:
BUILD_DIR = "axoloti_build" - axoloti dir/patch
RUNTIME_DIR = "axoloti_runtime" - axoloti dir
RELEASE_DIR = "axoloti_release" - defaults to axoloti dir

BUILD_DIR , where builds are done (needs to be writeable)
RUNTIME_DIR , this is stuff that doesn't change, basically the platform (read-only) and is quite a large download
RELEASE_DIR , the axoloti release... everything we want to release regularly in small download.

BUILD_DIR 
... is empty and will be filled with things like xpatch.*

RUNTIME contains
chibios
CMSIS
patch  (only the makefiles etc)
firmware *
platform_linux/mac/win

RELEASE contains:
application bundle, which will contain
App + JARs
patches (for demos etc)
objects
firmware *

Limitation:
- it needs testing on windows... it should be fine, but untested

Ive not updated the firmware scripts yet, this means that the firmware stuff is copied and built in RUNTIME_DIR, this needs to be changed do the builds are in BUILD_DIR and it is sourced from RELEASE_DIR. 

essentially to move firmware we need to start editing scripts which are platform specific, hence this is 'part 2' 

additional note: currently I build into BUILD_DIR, this will change to BUILD_DIR/patch and BUILD_DIR/firmware (or similar)

Note: as such this directory movement, is useful even with an application bundle, as it is a proper separation of deliverables and builds, with the correct read/write contexts.

TheTechnobear, 2015-07-13:
- new build.runtime target for mac and linux to build a runtime image, linux gets a tar file, mac gets a DMG
- new firmware environment var used to pickup firmware, defaults to release area, but can be overridden by user to build from somewhere else (axoloti home/firmware is good)
- axoloti_home replaces axoloti_build

to do
- store preferences in axoloti_home/preferences.xml
- create 'default firmware upload menu'
- test with an application bundle.
- compile/upload scripts for windows (in platform_win), copy pattern used by linux/osx
- build.runtime for windows... again copy linux/osx model
NOTE: build.xml deliberately uses native commands it HAS to to ensure file permissions are left intact, all ANT tasks will remove execute permissions. (and they are a pain to add back!)

TheTechnobear, 2015-07-13:
home is now defaulted to ~/Documents/axoloti  (mac) ~/axoloti (linux) HOMEPATH/Documents/axoloti (windows)  - if run without environment set... e.g. if you don't use the axoloti.sh etc
(i.e. this is for application run mode)

preferences now can be setup for runtime and firmware directories.

I guess we will need to default : 
- runtime  to something like Program Files/axoloti runtime)
- firmware/release based off the location the jar has been loaded from... this will give us an install directory.

still to do, the default firmware uploaded... I'm going to probably just do the by flash memory... we don't really want 3 different ways to be displayed for most users.

( we may need to consider putting the expert mode on preferences dialog ... just in case?)

TheTechnobear, 2015-07-13:
defaults runtime
decided not to use jar location for release, as we know in the application bundle it sets correct location as current directory.

default firmware flasher added

I think thats it!

I need to test with application bundle, but that should work.

TheTechnobear, 2015-07-14:
Mac, tested and OK
Im testing Linux - now

TheTechnobear, 2015-07-14:
Linux, corrected and now working :+1:

TheTechnobear, 2015-07-17:
is windows now done? ( I saw some checkins related to this)

vormplus, 2015-08-16:
@TheTechnobear: Guess this issue can be closed now.

TheTechnobear, 2015-08-16:
indeed, I was waiting on confirm of windows, but given we have released it... I guess its done :)

closed 2015-08-16 by TheTechnobear