axoloti/axoloti #97: GUI: Typing space in a comment object triggers the object selector

JohannesTaelman, 2015-07-12:


TheTechnobear, 2015-07-12:
it doesn't for me on OS X...  
its funny, I actually had worried this might be the case a couple of days ago, so tested :)
(and retested now)

does it do it for any of the other shortcuts  e.g N?

TheTechnobear, 2015-07-12:
got it... it appears to only do on edit.... gonna test some more

JohannesTaelman, 2015-07-12:
n triggers it too

TheTechnobear, 2015-07-12:
yup , quite serious as Ive put in quite a few more shortcuts on the editor... 
I'll take a look... I think we probably just have to take away 'focus' from the main panel

TheTechnobear, 2015-07-12:
BTW: comments contents should not be the object id... it means you cannot have 2 comments the same :)

JohannesTaelman, 2015-07-12:
also triggered when editing the instance name of an object

TheTechnobear, 2015-07-12:
fixed... its actually only N... 
its was caused by me adding an accelerator to the menu, and it appears the menu is still active when then editor is active.
for the time being Ive removed... since we don't have accelerators for the other panel shortcuts.

we/if we do, then I think we will need to check the component that is active at the time
but seems low priority for the moment.

commit 846eb470 (2015-07-12): Merge pull request #99 from TheTechnobear/unstable

remove accelerator fixes #97
commit c5cd0440 (2015-07-12): remove accelerator fixes #97
closed 2015-07-12 by JohannesTaelman