axoloti/axoloti #105: remove 808 samples from firmware

JohannesTaelman, 2015-07-13:
and introduce a better alternative: store on sdcard, and load into sdram

TheTechnobear, 2015-07-24:
just had an 'idea'

I definitely think samples should be on sdcard only and not in the firmware. 

BUT, my fear is this will kill many of the demos, which new users will expect to work.
but asking them to insert an sdcard and uploading samples is probably an unrealistic expectation...
 it also means sharing patches with samples is a bit of pain.

so the idea is this (and its really in two parts)

if a patch contains a sample reference it will, when going live:
1) check if sample is already on board (to stop continually uploading!)
2) if not on the board it will
a) upload to SDRAM 
b) store on SDCARD if the one is present.

the idea is, if you don't have an sdcard the demo will still run because it will be in memory.
(of course if you don't run from the computer it won't work)

also this provides some kind of 'automatic' uploading of samples, so you don't have to 'think about it'
(perhaps a CRC check, to check the sample hasn't changed)

(not a before release thing, but a later revision)

JohannesTaelman, 2015-07-24:
Yes.
I imagine an object field that supplies a library for inclusion to the linker. Samples or more general data may be compiled to a library - same fashion as the samples in firmware are. The linker needs to output a patch binary (that gets loaded into SRAM), and an SDRam binary image. The firmware can copy the sdram binary image from sdcard to sdram when starting the patch. Symbols locations will match. Memory can be allocated to fit the samples, as their size is known at link time.
More advanced, rather than copying the dataset in an sdram binary image (that would be regenerated every time a patch is compiled), only using the symbol locations from the linker, and loading the samples to the symbol locations at runtime.
Not a release feature, but important for a classic sampler use case, and more wild stuff too.
Then we can do a General Midi sampleset, with bagpipe, goblins and helicopter sounds :)

geslotencirkel, 2015-11-07:
Nooo, 808 samples in firmware is 1337.

commit 53da8a29 (2016-10-23): FW: remove 808 samples, closes #105
closed 2016-10-23 by JohannesTaelman