axoloti/axoloti #118: Object : attribute value

TheTechnobear, 2015-07-15:
in the karplus strong demo, we can see we need a constant that is derived from the delay buffer size.

it would be nice if this could be derived automatically.
my thought is to create a new 'attribute constant' which can be given a attribute name e.g. writebuffer.size and outputs this as an outlet.

possible, as the attribute essentially is a build time constant.

I think a simple object should work for most users... (AxoAttributeValue ?) 
so 
`AxoAttributeValue ref='instanceid.attribname'  outlet=value`

not sure if we will need difference types? need to check attribute types allowed.

(of course a more efficient and perhaps complimentary route, is to have a object which allows an expression to be entered, as then the maths would be compile time constant, and so more efficient in compilation (e.g. no runtime calculation)

TheTechnobear, 2015-07-15:
ok, I've noticed the attribute 'values' are hard code.
we should replace these with 'typed' constants
e.g. 
from 
` f = %fps% 
to
`const int ATTR_FPS =%fps%
`f = ATTR_FPS

will make code make much more sense too, when reading the generated xpatch.cpp

JohannesTaelman, 2015-07-15:
An objref attribute could be used in mtod. But I can also imagine use cases where it should compensate for two buffers. 
Attributes are really substitutions in code, allowing compiler optimizations. 
Have a look how table/read accesses constants of a table/alloc object for instance. Typed attributes will only work for specific attributes, not in general.

Perhaps a dedicated tuned delayline object, so the user can complement it with a filter? But also here, filters can introduce specific delays that need to be compensated...

A dedicated Karplus Strong object would be better since it can do higher-pitched notes, because it can escape the 16 samples buffer delay limitation.

TheTechnobear, 2015-07-15:
> Attributes are really substitutions in code, allowing compiler optimizations. 
constant variables/expressions will be just as optimised, as a constant bit of text .. gcc is pretty smart :)

I agree about the specific case, but I think generally all controls (params) and attributes should be some how accessible from other objects, wether its thru outlets of whatever.

from a user perspective, attributes are a pain... but I fully agree, they have a very useful place in code optimisation (and also due to array allocation at compile time)

so whilst there may be the odd case, where attributes are just used for 'code injection', I think for most cases, typed attributes, would be a good idea... and this would then allow for the kind of code mentioned above.

anyway, perhaps a 'save for later' ... when I raised the initial issue, id assumed they were compile constants with type... as there not, its a bigger changes.

JohannesTaelman, 2015-07-15:
Odd cases, well generic code injections seems very useful to me, like the script/script2 and midi/in/script objects show. It provides a nice thin wall in prototyping.
It would be nice if a script-type objects would also be able to declare its in- and outlets. One way to allow this is transitioning from in- and outlet declaration in XML to "plain" C++ files for objects with C++ types with doxygen-like annotations for in- and outlets. 

I agree about the addition of more formal attr subtypes in the long term.

Would you want "wired" ObjRefs? 
In PD named object arguments are used to reference tables and delaylines too.
If it 'd be wired, this 'd avoid the need for an object instancename. For 90% of the objects instancename is irrelevant, it only helps associating objects with generated code.

TheTechnobear, 2015-07-15:
perhaps 'odd cases' is the wrong phrase... but i means its a 'developer feature' rather than user feature.

script objects... yeah, I was thinking about these...
basically they start to overlap with the object editor...  what we want to do is to create a new object and then edit with the object editor... where we can use the attribs/params/inlet/outlets tabs to add new elements.
(perhaps with templated 'new objects' to allow for the looping and threading behaviour you have in the scripts)

but of course there is a question, about if the object editor could perhaps be made a bit more graphical, 
so you could drag in inlets/outlet/controls.

my guess is the first stage is just doing the object editor, which actually is not that much work really.
(though we might want to consider 'read only' attributes on the system objects... perhaps make users 'duplicate them' if they want to edit them and save as something different ?)

JohannesTaelman, 2015-07-15:
Script objects avoid the creation of a dedicated object that would be a one-off. 
You make a valid point, with a fully functional object editor, a dedicated object may also be stored in a patch for one-offs, just like a patch/patcher subpatch.

TheTechnobear, 2015-07-15:
going even more off topic (but related :))
Id like the ability to embed sub-patches within the main patch... (and similarly this one off objects could (optionally be embedded)

both max and reaktor have this concept, of either sub patches as separate files or embedded into the main patch.. its very convenient for objects you don't plan on reusing, as the patch is then self contained in one file.  (and of course you still have the option of later cut n paste the thing into a new patch and save it as a subpatch)

JohannesTaelman, 2015-07-15:
patch/patcher embeds sub-patches in a main patch already

TheTechnobear, 2015-07-15:
then why.. does it ask you to save... and create a file?

JohannesTaelman, 2015-07-15:
Ok created issue #121