axoloti/axoloti #120: OBJ:patch/patcher - issues

TheTechnobear, 2015-07-15:
a) request saving
when closing a patcher it asks users if they want to save.... and if they select yes, then it creates a file.

it shouldn't as it should be embedded within the main patch.

the issue is, its started in a normal patching windows, which just knows its a patch, and if its changed or not...
unfortunately there is no 'type' on patches, so the window cannot see its a special (embedded) patch that does not need saving
(we code potentially rescue 'subpatchmode?') 

b) update... the patch window should automatically update when inlet/outlets are changed.
(this implies using an observer model, so the main patch knows)

c) dirty on main patch
if a patcher is altered it does not mark main patch as dirty... so user can lose their changes

JohannesTaelman, 2015-07-15:
I agree,
one note about update, this can only happen when the parent patch is not live.

TheTechnobear, 2015-07-26:
committed change with address (a) and (c) 

(b) will come when we start link parent and subpatch frames