axoloti/axoloti #125: GUI: "File > Open”, File Format by default is "Axoloti Subpatch"?

JohannesTaelman, 2015-07-17:
Reportedly on OSX...
It's fine on Windows (defaulting to all Axoloti file types), though.

TheTechnobear, 2015-07-17:
I think it defaults to axoloti files on OS X too, but will check.
(Possible they were saving an axs ? It defaults to type loaded I think )

TheTechnobear, 2015-07-17:
(oops misread, as Save As ... which correctly defaults to patch)

anyway, just re-tested Open, and it correctly defaults to Axoloti Files (i.e. to allow all axoloti files to be opened)

TheTechnobear, 2015-07-17:
linux is also correct...
who said it was wrong? can we get some more info?

vormplus, 2015-07-18:
Only had the problem on Mac OSX 10.7.
Compiled on 10.9.4, and moved to my old machine via Airdrop.

JohannesTaelman, 2015-07-18:
So it is perhaps Apple JVM versus Oracle JVM and thus fixed in the bundle.

vormplus, 2015-07-18:
It's possible that my old machine is running Apple Java. Don't think I've installed anything else but Processing that required Java.

I'll test it again when the bundle is available.

TheTechnobear, 2015-07-18:
im transferring to a 10.7.5 machine now for testing...

TheTechnobear, 2015-07-18:
ok, the bundle works fine on my 10.7.5 box, because I have java 1.8 installed.
from the command line , I forced it to use java 1.6 (which comes with OS X 10.7.5) and yes I get the 'issue' 
it would appear between 1.6 uses the last file chooser added, verses 1.8 which uses the first.
unfortunately there is no way to explicitly choose.

we could, put in a hack for 1.6 to add these in reverse, but frankly I don't think its worth it,
as 1.8 is freely available for Mac OS X 10.7.5

generally, we are going to come across a lot of UI 'fixes'/differences between 1.6 and 1.8, and possible apple vs oracle vs openjdk... hacking around all of these will make the codebase overly complex, and frankly use limited development effort in an inefficiently.
(as theoretically we will have to start testing X platforms with Y JRE versions... it gets out of hand very quickly)

TheTechnobear, 2015-07-18:
note: we are not shipping the JRE with the bundles, as whilst its possible it will again, increase the size significantly.

vormplus, 2015-07-18:
Don't think there will be a lot of users on 10.7. If they still use that version of Mac OSX, they should install Java 1.8.

It might be a good idea to put Java 1.8 as a system requirement with the release notes.

TheTechnobear, 2015-07-18:
yeah, or at least 'recommended'... (it may be fine with 1.7 as well, but I don't have it, or time to test it)

actually I think  recommended platform swould be a good idea, and what I mean by this, is what we have tested against (and use frequently enough to spot any issues)
I''ll put something on the forum... what versions of Windows are you using?

JohannesTaelman, 2015-07-18:
I'm using win 7 and 8.

TheTechnobear, 2015-07-29:
close me :) 
(due to old java, which we won't support explicitly)

closed 2015-07-29 by JohannesTaelman