axoloti/axoloti #127: pasted objects - different size from original

TheTechnobear, 2015-07-19:
take a ctrl object, copy and paste

the new copy with be slightly smaller.

JohannesTaelman, 2015-07-19:
Also when opening a patch. 
This issue has been present for a long long time, and applies to all objects.

Object sizes are rounded up to grid, this often hides the issue.

In future, I think objects could specify their absolute layout rather than auto-layouting.

TheTechnobear, 2015-07-20:
agreed, I think in the future objects will need to have a size, and all attribute/params will need position info, this will allow custom layouts.

note: 
in the codegen we will need to 'auto layout' these, as doing by hand would be extremely tedious... 
OR 
we have a auto layout option/best fit option on objects so that if size/positions are not specified we still get a 'reasonable layout  (perhaps better)

closed 2016-08-19 by axoloti