axoloti/axoloti #129: remove transitions.xml

JohannesTaelman, 2015-07-19:


TheTechnobear, 2015-07-20:
do we need to replace with a different mechanism, or just assume zombie handling will allow users to easily update objects?

TheTechnobear, 2015-07-20:
Im wondering if replacement, could be the start of the upgrade process?
(also perhaps it would be better to do the inlet changes this way)

whats our thoughts on this...

my (only quick/initial) thought is, each object type could have an overridable upgrade method - upgrade(from, to)
when a patch is loaded it calls this for each object (assuming patch version!=current version), with patch version and current version.
upgrade methods would be written like
'''
if (patchversion<1.0) {
}
if (patchversion<1.1) {
{
'''
written so basically, an object may go thru multiple upgrades to go from 1.0 to 1.5)
to keep this from getting out of hand, we can then periodically, remove the lower patch upgrade parts... once we know everyone has transitioned. (i.e. we can end of life versions) ... though they would still be upgradable by using an older version of the software.

JohannesTaelman, 2015-07-20:
Obsolete objects may be an object type (updater object), loaded via the sha or uuid hash (without version number). It could contain a javascript that takes the old instance xml as input, and outputs upgraded xml. That would allow to:
* rename inlets/outlets
* adapt to new parameter scaling
* even interact with the user, show a wizard or something
When the output of that script emits another outdated object, this could trigger another updater object.

Why using a patch version number? I Object versions do not necessarily sync up to a patch version number. 

Javascript in objects may also be a way to deal with 
* parameterized object variations, for example mix/mix 7 could run the mix/mix javascript-object-generator, emitting a 7 channel mix object. 
* complex parameterized compile-time coefficient computation
one (advanced) use case is static (non-sweepable) filter design like Matlab "fdatool" through an interactive wizard, this sort of coefficient generation does not need to happen during runtime, can't be expressed in the C++ preprocessor.

Of coarse emitting object XML in javascript adds another layer of abstractness, for simple objects it is easier to edit the XML than editing javascript emitting XML. 
XML containing Javascript that generates C++ in XML, that 'd be a lot of nested character escaping... only manageable with generator code - yet another layer.

TheTechnobear, 2015-07-20:
yeah I think we need to take care here...
adding yet another programming language always reduces the number who are confident to work on the project..  (we already require quite a rich set of skills,  C, Java, XML,shell scripting, ant etc )

Also, I think we need to be really careful with this code gen direction....

one of axoloti strengths is you can write an AXO by putting some (potentially simple) C in an XML file, 
this is cool and then it holds a descriptor for inlets/outlets params.
I think we need to take care we don't start imposing the java code gen on other axo developers

TheTechnobear, 2015-07-20:
whilst walking the dogs, I was musing over this a little...

the key thing about an 'upgrade manager' , is you do **not** want to pollute your current code base/objects with anything to do with past versions, or the upgrading process.
(the code is complex enough, without this extra 'noise')

so you really want the upgrade code, to be separate, you also want this so that it can be 'pruned'.. i.e get to version 10, and you do not need the code for upgrading from 1 to 3 anymore, as most will have already upgraded. (and you don't want to have to potentially maintain this)

this led me to an 'off the wall' idea -  perhaps we could do upgrades by just directly adjusting the xml, rather going thru the object layer - after all we only need to upgrade patches, not axo's, and their xml model is not that complex.
so the way we could do the upgrade, is before we deserialize patch into the java object model.

(I guess an analogy, is the way often if you use a database, you run upgrades scripts on the database rather than doing it through your C++/Java object model)

as i say, its an off the wall idea... I thought might be worth considering...

patch version vs object version... 
yeah I know, actually I view 'patch version' as application/release version.
theres a few reasons I'm concerned about using object level versioning.
- it could become pretty onerous very quickly... with hundreds of objects having their own versions, mixing in different patches.
- when different objects interact, they could require to be on particular versions to work together.
e.g. i might add a new outlet on one object to be fed into a new inlet on another
- sometimes the upgrade might need to touch multiple objects 
e.g. above example, might want to create a connection between these 2 objects,
or we might decide to split one object into two objects, so the upgrade would have to delete the original object, and put the two new objects in place and then connect them together...

the above might seem unlikely (possibly is)  but i think a good upgrade process, would not assume too much... and whilst the upgrade process may not be simple to write, it would be possible

anyway just a few thoughts...

TheTechnobear, 2015-07-21:
hmm,  a couple of things before we axe transitions manager
a) we should make it so that, once a transition is made... the object/patch is updated, such that saving the patch automatically is with the new versions of objects etc. (currently it doesnt seem to do this)
(the way synonyms work, this already happens for them)

b) I need a way to go thru all patches (including subpatches), loading them and then saving them...
(quite similar to what test does, but saving)
this way, we can upgrade all the patches/help with the new names and synonyms without loading and saving each manually...

this is then like a very basic 'upgrade' feature... at least for our own maintained patches/help files.

(its possible I could perhaps put simple object mapping into the synonyms )

JohannesTaelman, 2015-07-21:
fully agree with a) and b)
also notice that I sneaked in another upgrade mechanism in objects - "upgradeSha", to preserve sha references when transforming %in% into inlet_in etc.

For future object upgrading I believe both upgrade references (upgradeSHA) in objects and a central database (transitions.xml) are not the right approach.

TheTechnobear, 2015-07-21:
how are you planning on using the Sha?

the issue I see, is that a user only sees an object (type) name, and can accept that the same type has different inlet/outlet types...
so when upgrading, they will expect that a object with the same name has to be functionally compatible with the previous version... they are not in a position to make a judgement about upgrading, this is the developers responsibility.
(the user will also be used to using obj X, and it doing X... so even if you could rename during an upgrade this would be bad for creating new patches, as the users knowledge is now out of date)

even object identity is a bit interesting...
if Ive got an object X inlet a,b... and object X inlet a,b,c then if 'c' is optional, as far as the user is concerned these objects are interchangeable.

so basically when upgrading, id be happy to take a new object if it has the same name (so I assume the same function) as long as it has all the inlets/outlets i was using, and contains at least the same params/attribs as before... in both cases more inlets/outlets/params/attribs is not an issue.

the issue is I don't think you can 'program' that by calculating a SHA, which can only express its equal, not that an object is a super set of the original.  
and.. I think that is the most common case when upgrading, you tend to extend objects, and we don't want to have to give a new name each time, we will run out of names fast :)

JohannesTaelman, 2015-07-21:
The motivation behind the SHA tag is that it proofs a 100% identical object as used when creating the patch.

What is considered an upgraded object, could also break patches, as soon as the upgraded object produces any different output.

It is not always easy to be sure what is an upgrade. For example, a bandwidth limited saw oscillator is a tradeoff between accuracy and computational cost, and different strategies are possible. Even when an improved saw oscillator with better accuracy at less computational cost is added, some patches may get their specific sound from the older saw oscillator.

Eventually I hope to keep the possibility to open patches using its original objects. Bug-for-bug compatible. 

While dragging along bugs forever is annoying. The UUID (object unique identifier) tag is intended to stay identical when upgrading objects (currently as UUID is generated in codegen), and unless patches are opened in a bug-by-bug compatible mode, they could upgrade objects by UUID. 

Extended objects (with additional parameters, inlets, outlets, attributes) can live by the same name, different UUID (and SHA). 

Keeping the growing forest of objects under control is a concern, yes. When objects are only standard math library wrappers their lifetime is long because they have very strict definitions. But standard math library is optimized at uncompromised accuracy, in dsp accuracy/speed tradeoffs are common. Dsp concepts also have strict well defined concepts, but exposing that primitive level will make it incomprehensible for many people.

*This is what I had in mind for upgrading, but I'm open for arguments:*
 
Upgrading an object involves:
* moving the obsolete .axo to an "obsolete objects" folder, renaming the object file to its SHA hash
this will become a wild forest
* adding an .axo object in the "objects" folder that has the same UUID hash so documents will get upgraded unless the obsolete objects are loaded.

Alternatively, when the upgraded object is not a drop-in replacement:
* first step same as above
* add a "magic upgrade object" with the same UUID, that describes the patch adaptations needed to make it drop-in compatible, and forwards to a new UUID. Javascript-in-xml perhaps.
* adding an .axo object in the "objects" folder with the new UUID

Renaming an object - to clean up the namespace, can be just renaming the object, patches will find the object back by UUID. I agree that renaming must be minimized.

TheTechnobear, 2015-07-21:
k, so your ideas is...

that when a user opens the patch after an upgrade (thats has updated objects), it is automatically using a newer versions   
HOWEVER, there is a mode where they can open 'as it was before upgrade' i.e. using the obsolete version folder.  - correct?

principle being they can get the patch to sound identical to how it did before... correct?
(e.g. a bug may have been fixed, but they might in that case, still prefer the sound of how it was before.. or may have a gig tommorow, so just want what they already know)

I do kind of understand that... its important for musicians to be able to go back...

my fear is this could end up requiring quite a bit of functionality to be useful... 
e.g. they are bound to want to start editing it, and cut n paste... all the time the editor has to be careful to use either a legacy or new object... and god forbid they want to search to get the old version of another object.

then there is the fear of mixing old and new versions, i.e. you create a new version of an object, which is already loaded as a legacy object.. how can the user distinguish.

and then just the general issue, how does a user know what the difference is between the legacy object and its newer counter... especially as there might be 3 different legacy versions...

so yeah, I get the I want it to sound as it did before argument - that is cool 
BUT.. I just don't know where they go from there... even if we indicate on the UI, its a legacy version, that still tells them very very little.

Is this much better than just keeping an old copy of the patches/objects as well as the upgraded version in an archive folder?

JohannesTaelman, 2015-07-21:
I think quite some functionality is already there.

Obsolete objects should not be visible in the object selector, and not preloaded at startup. SHA tag as filename? 

Obsolete objects would only be instantiated by
* opening a patch in "as it was before" mode
* copying objects from there into another patch...

Obsolete objects may be made recognizable in a patch by showing their SHA hash, below the object name or something (ouch this breaks visual compatibility already). The SHA tag is humanly distinguishable, while also annoying, and that could motivate upgrading to the current canonical object :)

> Is this much better than just keeping an old copy of the patches/objects as well as the upgraded version in an archive folder?
Well when someone decides to add that archive folder to his object search patch, trying to open a patch "as before", all multiples will be visible in his library, indistinguishable.

TheTechnobear, 2015-07-22:
still not totally convinced :) 

>Well when someone decides to add that archive folder to his object search patch, trying to open a patch >"as before", all multiples will be visible in his library, indistinguishable.

this is not quite true, object path is relative, and objects are bundled with app... so if they run an old version of an app they will get the object it came with.
(so works much the same most apps, you can run an upgraded version of a patch... or run the old app with the original patch)

ah, I hear you saying "but what if I run from github" ... then use versioning it supplies, no reason for us to do a 'limited' version of what it can do  (we will be tagging releases anyway, and users can also tag their own repos too)

also from a implementation view... using your approach:
you will have to have a copy of every object that changes (and each time), and this will have to be in 'the user area' (as it needs to be writeable by the app)... because you have no idea what objects they may have used in the patch. 
thats a lot of objects, the vast majority, of which are not being used by anyone.

also I still have some really bad feelings about this mixing...
- the SHA is not a way for a user to understand the version he is using, yes he can see its different, but it tells him nothing about when the object is from, or why its different. (date arguably is useful)
- the idea that someone could paste an old version of an object into a new patch, is asking for trouble, it means old bugs will proliferate into new patches.
(people will copy bits of code they have working from one place to another, we all do this, all the time)

do get me wrong, I do understand the requirement behind it, but Im really concerned about this creating a support nightmare, people expect 'legacy objects' to be supported, complaining that their old patch has stopped working (or is behaving differently) .
lets be clear, we are not a commercial company.. no open source developer will be interested in helping out someone who's old patch doesn't work, they will just tell them to upgrade, to move forward. we just don't have the resources to be anything other than forward looking.

finally, I really don't like this idea that we need computer generated ids in the axo, as it means this cannot be crafted by hand... or at least we have to provide a program that post processes them.
(and as mentioned earlier, I think its a very bad idea, that this XML files **have** to generated in a java program... I don't like it generally, but don't mind if its optional, thats the developers choice)

so heres my take... 
users can keep old copies of the app. patches when upgraded, keep a copy of original version (good idea anyway) .. you want the original use the original patch, original app.  
(its a model users are very familiar with)

I do have another suggestion.  a user should be able to store the old 'binary' patch, so they can upload it.
I think this could either be (or perhaps both in some kind of archive) 
- the generated xpatch.cpp  ... this would allow them to use against a newer firmware (assuming its still compatible, which is pretty likely)
- the binary firmware.. this could only be used if the firmware has not changed, or we store the firmware with it.
I think this feature, would be really useful... and for most musicians is the one they would use most.
(i.e. they played a gig last year, and they want to play it again... but don't have the time to start upgrading the patch etc)

so I guess my view is the (normal) users has the following options:
- upgrade patch, run new version
- run old app with old patch 
- use an 'archived' image 

... what they don't have the option to do, is to use old objects in new patches, or mixing old and new.

so I respect yours is a more flexible approach, but not one id use, with this in mind, if you follow your route can I ask you ensure:
- vanilla axo can be used without version, without glaring warnings.
- vanilla axo can be written in XML, without any post processing.
- optionally, I don't have to store legacy objects... i.e. its a user choice, that they always just upgrade
- none of this, complicates the main code line (which is complicated enough without having to deal with different versions of object.... I think as you say this is largely the case already)

from my part, I think I will add the 'archive image' option, as I think this could be really useful :)

JohannesTaelman, 2015-07-22:
I understand your concerns about a potential support/maintenance nightmare. 

Some clarifications:
* Code-generated objects were a method to bootstrap the object library. At some point it will become obsolete. 
* I certainly agree axo objects should be creatable in any editor. 
* computer-generated SHA in axo could be left out, computed when loading. Or ask to add/correct the SHA in the axo file when loading. Or leave it blank, in that case there will be just no traceability. The motivation to put it in the axo file, is that you can find the matching object with a simple text search.
* uuid currently computer-generated, but really is just an unique identifier, to preserve association to moved/renamed objects and to preserve association with the right variant of same-name objects.

> also from a implementation view... using your approach:
> you will have to have a copy of every object that changes (and each time), and this will have to be in 'the user area' (as it needs to be [...]

Why would it need to be writable by the app?

The full catalog of official but legacy objects may also be stored on the server. 

But the main issue is more like individual object versions versus a global package version of Axoloti. My motivation for individual object versions is that a global package version does not tell me if an object has changed, or just the same with an upped version number.

I believe both package version and object version make sense, and can co-exist.

Once I imagined some sort of version control at object level, server side. So anyone can fork any object or patch without forking the whole repo. But then converging rather than diverging into a million fragments where no one can find the gold anymore is an enormous challenge.

TheTechnobear, 2015-07-22:
store on server - don't assume online access, many musicians don't have their studio machines connected to the internet.  they copy things from one place to another (I know archaic but you will see this argument repeated in many places)

the issue with object versioning is its not enough on its own... basically a patch is a collection of objects all at a particular version... swap anyone one object with a different version and it may (or may not) break/change behaviour. (e.g. if you upgrade an object and it sends another object a slightly different scales signal, it will sound different... even though it has not changes itself).

yeah, global vs object/patch version is only about simplicity, the former is simpler the later is more flexible.

the thing is,  do you really want to be writing a version control system, or music software...
if you want version control, id seriously consider using something that already exists... what can we add, except perhaps tighter integration?

anyways, not a big deal... if its something you think is a priority, and worth the dev/support effort thats cool. Im not saying it cant/wont work, I'm sure it will.. and it will be cool. Im just not sure about the effort vs reward pay off... (hence my suggestions of 'lesser' solutions with slightly less long term effort)

TheTechnobear, 2015-07-22:
ok, -runUpgrades <directory name> will upgrade all axp,axh,axs it finds 
... works a treat, just done it on all patches etc.

so WE don't need transitions anymore (or synonyms) but we may need to leave it in there, so that existing prototype users can upgrade patches.  

(Id assume prototype users have limited expectations of compatibility with production units, as it was still in development - I think asking them to upgrade quickly, so we can remove the transitions/synonyms is reasonable)

JohannesTaelman, 2015-07-22:
No, implementing the loading "as before" and update scheme, or integrating version control at object level is not a priority. 

Dealing with these issues you mention is a priority where they apply to the current state
1. vanilla axo can be used without version, without glaring warnings.
2. vanilla axo can be written in XML, without any post processing. 
3. optionally, I don't have to store legacy objects... i.e. its a user choice, that they always just upgrade
4. none of this, complicates the main code line (which is complicated enough without having to deal with different versions of object.... I think as you say this is largely the case already)
where needed.

I believe 1 and 2 deserve a separate issue.

runUpgrades: awesome!