axoloti/axoloti #138: OBJ: fix tests

TheTechnobear, 2015-07-20:
first results of all tests run:
need to go through each and either fix or raise bug which prevents fixing...
(close when either FIXED, or another issue is raised, as that can then track the issue)

SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/demos/keyboard/organ.axp
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/demos/sequencing/drseq.axp
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/demos/sequencing/groove.axp
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/demos/sequencing/scripted_midi_drama.axp
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/demos/sequencing/sequence.axp
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/demos/youtube/guitar_multi_fx.axp
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/demos/youtube/H2SO4.axp
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/demos/youtube/soundtrack/bass_lead_drum_trio2.axp
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/tutorials/15_waveshaper.axp
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/tutorials/21_modulated_delay1.axp
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/tutorials/22_overlap_add_shifter.axp
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/objects/delay/echo.axh
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/objects/lfo/taptempo.axh
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/objects/midi/in/script.axh
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/objects/patch/control_on_parent.axh
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/objects/patch/subpatch_polyindex.axh
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/objects/patch/subpatch_polyindex_sub.axh
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/objects/patch/zombie.axh
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/objects/unstable/spectral/vocoder 24.axh
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/objects/wave/wave.axh

TheTechnobear, 2015-07-20:
I'll edit this as I work on them:
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/demos/keyboard/organ.axp
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/demos/sequencing/drseq.axp
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/demos/sequencing/groove.axp
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/demos/sequencing/scripted_midi_drama.axp
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/demos/sequencing/sequence.axp
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/demos/youtube/guitar_multi_fx.axp
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/demos/youtube/H2SO4.axp
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/demos/youtube/soundtrack/bass_lead_drum_trio2.axp
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/tutorials/15_waveshaper.axp
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/tutorials/21_modulated_delay1.axp
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/tutorials/22_overlap_add_shifter.axp
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/objects/delay/echo.axh
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/objects/lfo/taptempo.axh
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/objects/midi/in/script.axh
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/objects/patch/control_on_parent.axh
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/objects/patch/subpatch_polyindex.axh
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/objects/patch/subpatch_polyindex_sub.axh
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/objects/patch/zombie.axh
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/objects/unstable/spectral/vocoder 24.axh
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/objects/wave/wave.axh

JohannesTaelman, 2015-07-20:
I'll create the missing "env/d m" object #135

TheTechnobear, 2015-07-20:
moved missing objects to #140 
re-running tests... if this passes will close this subject to #140 being fixed.

TheTechnobear, 2015-07-20:
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/demos/youtube/guitar_multi_fx.axp
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/tutorials/15_waveshaper.axp
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/tutorials/21_modulated_delay1.axp
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/patches/tutorials/22_overlap_add_shifter.axp
** SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/objects/lfo/taptempo.axh **
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/objects/unstable/spectral/vocoder 24.axh
SEVERE: COMPILE FAILED: /Users/kodiak/Projects/Music/axoloti/objects/wave/wave.axh

all reported in #140 except taptempo which I'm checking now...

TheTechnobear, 2015-07-20:
tap tempo.axo broken due to undeclared variables raised #141 

so #140 and #141 reflect all issues known to stop tests compiling.
once 140/141 are fixed, we should re-run tests to verify.

closed 2015-07-20 by TheTechnobear