axoloti/axoloti #140: OBJ: missing objects

TheTechnobear, 2015-07-20:
the following objects appear to be missing, either we create or delete demo/help etc (or adapt them)

switch/tape delay/reverb in : demos/youtube/guitar_multi_fx.axp

test signal generator , required for 3 tutorials:
15_waveshaper.axp, 21_modulated_delay1.axp, 22_overlap_add_shifter.axp

the following do not compile, due to 'unstable'  ? should we remove, and if so which objects?
fail to compile : /Users/kodiak/Projects/Music/axoloti/objects/unstable/spectral/vocoder 24.axh
fail to compile : /Users/kodiak/Projects/Music/axoloti/objects/wave/wave.axh

JohannesTaelman, 2015-07-20:
working on this

TheTechnobear, 2015-07-20:
added a test signal generator... but really only works for sine/noise/saw... not quite sure how the sample playback worked before. ... but at least compiles :)

JohannesTaelman, 2015-07-20:
closing, samples relate to issue #105

closed 2015-07-20 by JohannesTaelman