axoloti/axoloti #148: precomiled patch header

JohannesTaelman, 2015-07-21:
Using a precompiled header for everything included in xpatch.cpp may speed up patch compilation

TheTechnobear, 2015-07-29:
I'll take a look at this, this morning

TheTechnobear, 2015-07-29:
ok, firmware/xpatch.h is a header file that contains the firmware includes used by patch
this builds precompiled into build/xpatch.h.gch whenever the files in xpatch.h change or xpatch.h changes  (log indicate its being built)
(so first Live patch will build it, but subsequent not)

we can tell its being used, by the logs when compiling.
! /Users/kodiak/Documents/axoloti/build/xpatch.h.gch

JohannesTaelman, 2015-07-31:
Precompiled header should be deleted when compiling firmware, otherwise weird things happen...

TheTechnobear, 2015-07-31:
how can that be? the precompiled header is xpatch.h.gch, so will only be used when compiling the xpatch which includes xpatch.h....
what could be 'odd' is if the firmware header files change, as i only check the status of the 3 header files I include, not the hundreds that are included from those...

are you saying the xpatch, goes weird?

TheTechnobear, 2015-07-31:
do you think this is some how related to the windows firmware build failures, if so I think (but not sure) thats something to do with the .dep, which is gcc dependency files.... which I do think should not be included in the release.

JohannesTaelman, 2015-07-31:
As an example: the definition of LED1PIN depends on the selected board in axoloti_defines.h
Switching to a new board profile, compiling and flashing, leaves the old pin definition active. A LED blink test then causes a different pin with a different function associated to "blink".

TheTechnobear, 2015-07-31:
is this due to code that is being run in xpatch... i.e. is not compiled in the firmware, if so, then that is expected ... we could add axoloti_defines.h to the dependancies on xpatch.h.gch...  if that is one that is likely to change.

TheTechnobear, 2015-07-31:
(of course deleting the precompiled header on a firmware build is also trivial, but would be good to know why the issue exists.... )

JohannesTaelman, 2015-07-31:
I think the "lifetime" of the precompiled header exactly matches the firmware build lifetime. There are many other definitions that are potentially relevant and can cause confusion.
I think also the patch "ramlink" linker script and the makefile rather belong in firmware I feel. That does not make firmware modular, but at least fully contained.

TheTechnobear, 2015-07-31:
Yeah, I was also think the compile-patch/firmware should be in the app
bundle, not the platform
 as these kind of changes can impact them.
( or perhaps they call another script in firmware rather than hitting the
make file directly)
 Eg deleting the gch would be done in the scripts!

JohannesTaelman, 2015-07-31:
When using non-bundled but custom firmware, linkerscript, makefile (compiler flags), and precompiled header are likely also custom.

TheTechnobear, 2015-07-31:
ok, done... 
only thing I've not done yet, is move the compile_patch/compile_firmware

TheTechnobear, 2015-07-31:
ok done... compile scripts are now in firmware, which means they can easily be changed with app release.
the platform scripts merely set the environment and then call them.

java can now call the firmware/compile directly (to do)

TheTechnobear, 2015-08-16:
this can be closed

commit 4a36ad0b (2015-07-29): UI:use precompiled headers closes #148
commit 5ce8345f (2015-07-29): Merge pull request #165 from TheTechnobear/master

UI:use precompiled headers closes #148
closed 2015-08-16 by JohannesTaelman