axoloti/axoloti #151: GUI: presentation view

TheTechnobear, 2015-07-22:
Patch/UI Separation... presentation view.
currently the UI is embedded in the patch. which means you need the original patch to run it.
(e.g. so if you archive a patch  the axoloti board could run it, we couldn't have any UI interaction as we don't have a UI without the AXP)
so it would be beneficial to have a UI representation that could be stored alongside the firmware that could be used without reference to the AXP.

This would appear to be a significant overhead, but I think it actually might be a useful feature anyway 
- On a computer, you in the future the UI will be reconfigurable, i.e. you will want to be able to layout controls completely separately to the patch/object layout (like Max, presentation mode and Reaktor)
... potentially a much simpler application could be used for this presentation view, which would link really nicely to archiving patches.
- for hardware devices, they already use an inspection approach (i.e. they don't/cant use the patch), again we may want a bit of flexibility here. e.g. favourite controls for a patch.
(perhaps this can be related to the presentation view above, but you can save different views for different devices e.g. one for laptop, one for axoloti core, one for your iPhone)
- mobile device control.... many use iPads etc, we don't need full patch building on an iPhone, but it would be nice to be able to control the patch from one.

JohannesTaelman, 2015-07-22:
A compiled patch declares its own parameters for embedded interaction. This has been developed for Axoloti Control. Check Window->Remote in Axoloti, you get an interactive virtualized remote interface. When loading a patch, you can enter the "patch" submenu, and tweak patch parameters. It's certainly not bug free. And it is made with a minimal B&W screen + knobs interface in mind, but the bitmap rendering is done in firmware not in patch. It does not have coordinates yet.

Adding a protocol to retrieve the same patch parameters+metadata through USB in firmware sounds very doable and useful. The overhead is in firmware, and as long as skinning, background pictures etc, is not required, very small.

Not sure it fully aligns with presentation mode. Different devices have wildly different needs. You can't obey presentation mode coordinates with a knobs+display device. Upscaling a 128x64 monochrome bitmap to an iPad is also not acceptable. When there are more parameters than can fit on the iPad screen, I'd prefer organized pages rather than scrolling a virtual canvas. A VST or audiounits interface often tries to minimize screen space to keep the sequencer visible, while on iPad you probably want to use the full screen. 
Presentation mode is certainly on the map, first step is to reflect it to the parent object in a patch. Control/display coordinates may then be exported to the embedded parameter data structure too. 
But let's "save for later", ok?

TheTechnobear, 2015-07-22:
yeah, I really don't think bitmaps are the way to go... its too limiting for all devices.
and as you say, different devices will want different layouts.
( we actually do this in EigenD, it saves different layouts of controls for different resolutions)

yeah I think first step would be transmitting the meta data about parameters  (and values) over USB, this could then potentially be interpreted 
(pity we cant use OSC here, would be pretty ideal for using with things like Lemur)

yup save for later... partly this is fleshing ideas out regarding where we are heading...

JohannesTaelman, 2015-07-22:
offtopic about OSC
OSC implies an IP network, right? Wired ethernet, while very powerful (AVB standard etc), is also a bit annoying to deal with plug and play, dhcp etc.
A deamon app could bridge the native bulk Axoloti USB interface to localhost OSC.
Or there are (several) standards to tunnel IP over USB. That way OSC can be done without deamon or drivers, but far more work, that is not on my map.

TheTechnobear, 2015-07-22:
apps only support it over IP, so yes, a daemon forwarding would be the solution.
not difficult once we have an 'open protocol' for describing the controls and the data.