axoloti/axoloti #152: Running legacy patches

TheTechnobear, 2015-07-22:
(originally discussed in #129, see that for backgound)

Use case: Musician wants to play a session from a gig last year, he wants it to be exactly the same.
(rather than using an upgraded version of axoloti and the patch... e.g. due to time constraints, or for re-recording) 

here are some solutions:  (from #129, more than one might be applicable)

Solution 1:
we allow archiving of an image, this includes
- firmware binary and compiled patch
- generated code (xpatch.cpp etc) 
- binary UI (see ideas on presentation view #151)

Solution 2: (I don't like this) 
we archive all source into one file, this includes 
- patch files, and all objects used
- the editor then is able to open this archive, using this archive as the source for all objects.
- (optionally, store firmware as might have changed)

Solution 3: 
- when upgraded old patches are saved, and user can run with old version of app.

Solution 4: 
- when upgrading objects, we store the old objects in obsolete_objects and can load these via SHA key if in legacy mode.  (more details in #129)

JohannesTaelman, 2015-07-22:
I just add the downside for every solution here:

Solution 1:
* can't run on any different hardware

Solution 2:
* does not survive api upgrades, will need firmware source too

Solution 3:
* you mean ~~upgraded~~ old patch?

Solution 4:
* cfr #129 

And one extra:
Solution 5:
Export the patch as a single .axo object
* no need to maintain patch api backwards compatibility at c++ level, only .axo backwards compatibility.
* where objects address firmware, this potentially breaks the .axo

TheTechnobear, 2015-07-22:
Solution 1... ? 
you mean, you can just port it to say, an axoloti v2 board? I don't see this as an issue, if your using different hardware, it potentially will behave different... you'd be better using the upgraded version.
(note: potentially if old objects may not work on new hardware/firmware?)

Solution 2... thats why i suggesting storing firmware binary too, (this has same hardware restriction)

Solution 3 ....yes old app with old patch

Solution 4 .. yup, I know splitting issue up, but #129 is already pretty off topic :) 

Solution 5 ... interesting, this is close to solution 1, with the xpatch? but wrapped as an AXO so it can contains UI elements?  Id worry this sounds like an alternative code generation routing, but honestly I'm not quite sure how you are planning on merging all the objects into one.

Note: Firmware, Im kind of assuming that the firmware will remain to some extent stable.
It should be noted, 4, will also have issues with running old objects on new firmware.
e.g. say we start sending midi as signals rather than thru the midi function, potentially only newer objects will work with that firmware... 
(again, objects are not the only thing that can change... patch formats, firmware, UI operation, all can have behavioural or compatibly changes... its why when you 'release' software, you tag it all, not just individual files)

JohannesTaelman, 2015-07-23:
Solution 1:
Yes that is an inherent limitation, and not really an issue. The only way out would be virtualizing the hardware :)
I could imagine a library of outdated firmware images on sdcard, trying to load an binary patch when a matching firmware is available could flash back old firmware. The downside: flashing is slow, and only specified up to 10000 erase cycles. When flashing every minute, this is 7 days of flashing. Unlikely to reach this when there is some manual trigger involved. 
The CRC32 firmware hash will give sufficient traceability from patch to firmware.
Certainly an acceptable option. Bug for bug compatible. Flashing a firmware from sdcard was implemented some time ago, but then upgraded to use sdram as firmware image storage buffer.

Solution 2:
Will also need firmware source code too, or perhaps a more minimal alternative, precompiled header #148 and firmware map file. I think solution 3 is a more complete version of solution 2.

Solution 3:
Always a possibility, but only a last resort I hope. My fear is users will just stay with an old trusted version, rather than doing any upgrade. And keep making new work with the old version.

Solution 4:
Will fail when objects interact with firmware api that is changed, this is most likely to happen with I/O objects, but unlikely with dsp algo's. Open-as-before will need to be selective and the I/O interface objects would need to assume the same scalings.

Solution 5:
wrapping a patch into an .axo is already there, and that includes all the generated source code of objects and sub-patches. 
The differences between an xpatch.cpp generated from a patch and an xpatch.cpp generated from an AXO generated from a patch is the addition of "root" presets, parameter modulation table, and root references to parameter exchange and display structures will be left empty. This scenario it is some sort of "everything on parent", but modulation-table-on-parent and preset-table-on-parent are weird concepts... mmm.  Compared to 4, I/O objects will also be wrapped in subpatch, a selective update of just those is not possible anymore.. Perhaps not so useful until the API has converged more.