axoloti/axoloti #155: GUI: delete key should delete selection connection

TheTechnobear, 2015-07-23:
only Q is, mouse selection appears to select entire network...
which means delete should delete network.

however, I'm not sure selecting network is the right thing either,  useful perhaps for visualisation, but not if we wish to allow connection manipulations (like delete)

JohannesTaelman, 2015-07-23:
notes
* Wires are currently not selectable, only highlighted when hovering over an inlet/outlet, this is useful to understand a spaghetti
* Wires themselves are not clickable. It gets in the way of manipulating dials (the eurorack problem, knobs behind patch cable spaghetti). I also fear performance of a mouse-hit-detector on curved wires. I think this can stay as is.
* I can see not much other use of selecting a net except deleting it. Copy pasting a net without objects would be weird. Group-selecting nets by dragging a rectangle around would be weird too, if it selects the objects + nets in the rectangle, only those will be expected in the past, while this would not be useful.
So a selected wire looks like a special case of selection.
Or would it be focus rather than selection? Hmmm tab-tab-tab now cycles through parameters/attributes nicely, if inlets/outlets would be cycled through too this is annoying.

I support the idea of delete key to delete a net.
Focusing through inlets+outlets seems more appropriate than selecting nets, but should be a different focus cycle than parameters/attributes.
There is currently no visual pattern for a focused inlet or outlet, this will be needed.
It may also extend to patching nets by keyboard. Focus outlet, press shortcut-key for "start net", tabtabtab to inlet, press shortcut key for "attach".

TheTechnobear, 2015-07-23:
Save for later :)

TheTechnobear, 2015-07-24:
yeah, I think wire selection is more natural via mouse... I've seen keyboard interfacing for connecting wires, and I never use it (u-he bazille) as its clumsy.

I think id save tab (and shift-tab) for cycling between objects, which has a much more useful set of operations. (replace, delete, edit object definition etc)

JohannesTaelman, 2015-07-24:
Tab/shift-tab for cycling between objects is also not implemented.
Tab/shift-tab currently cycles between parameters and attributes only. Useful for tweaking.
I believe multiple independent cycle chains could be made, so when an object is focused, it cycles between objects, when a parameter is focused it cycles through parameters, when an in/outlet is focused it cycles between in/outlets. 
But this may violate UX rules and is certainly not a priority.