axoloti/axoloti #169: DOC: pull html or markdown from community site into doc folder

JohannesTaelman, 2015-07-29:
curl'ing discourse?

TheTechnobear, 2015-07-29:
k, I think we only need the main user guide page, so thats easy enough... I can do that now.
but for display, I think it would be ideal if we could show in the java app....

there are a few markdown processors, that will process it, and turn it into HTML, which we can then display in java with ease...

Ive no experience, but this one looks ok:
https://github.com/rjeschke/txtmark
uses apache2 license, is this ok... or do you want to find something else thats more suitable?

JohannesTaelman, 2015-07-29:
Why showing it in the app when the system browser can be asked do open it?
I think raw html can also be curl'ed from discourse.

TheTechnobear, 2015-07-29:
yeah if its HTML then sure we don't need to render it (only if its markdown, as users wont have a viewer)
but Im not sure how we can get the raw html... any clues?

JohannesTaelman, 2015-07-29:
https://github.com/discourse/discourse/blob/master/config/routes.rb#L472
is the closest clue I could find

TheTechnobear, 2015-07-29:
cant see anything there... and don't really want to be running ruby.

JohannesTaelman, 2015-07-29:
no don't think you need to run ruby

similar to 
http://community.axoloti.com/raw/50/1
is a get "raw/:topic_id(/:post_number)" => "posts#markdown_num"

that works for raw markdown

get "/posts/:id/cooked" => "posts#cooked"
should return html
but I don't know how to make an id

TheTechnobear, 2015-07-29:
Id is 50 in this case

TheTechnobear, 2015-07-29:
ok, Ive checked in solution ... not sure if you'll like to much, but seems best for the moment.
in the ant bundle task, I download the markdown version, and then use a perl script to convert it to html
means the builder needs to have curl and perl installed... 
curl has to be for build tools
perl is pre-installed for linux/mac osx

so only windows users 
Note: of course this is only when you are bundling the app, not just build firmware/jar etc

TheTechnobear, 2015-07-29:
tested, and working on linux and mac osx.
(need just to confirm on travis, but no reason it should fail)

p.s. currently i check in (manually) user guide as well into github, 
so if windows does not have perl installed, it will fail, but continue and then use the github version of the user guide.html

closed 2015-08-01 by JohannesTaelman