axoloti/axoloti #170: Firmware ID = Please compile the firmware first

vormplus, 2015-07-29:
Found in 1.0pre13.

Menu item Board > Firmware only shows Flash and Flash (Rescue).
Compile Firmware menu item is missing.

I've tried compiling the firmware with the shell script found in:
/Applications/axoloti_runtime/platform_osx/compile_firmware.sh

This is the output I get in Terminal:

	Cloudswimmers:platform_osx janvantomme$ ./compile_firmware.sh 
	./compile_firmware.sh: line 12: cd: /Applications/axoloti_runtime/platform_osx/../firmware: No such file or directory
	Cloudswimmers:platform_osx janvantomme$ 

Looks like the firmware is not included in the axoloti_runtime download.

TheTechnobear, 2015-07-29:
yes, its there, but it needs to be compiled on travis.... then it will be
in the included and you will not have to compile it.
only users who use custom firmware will need this, and they can do this
from within the app.

its possible from the command line, even if its just an app install (rather
than github)  but thats 'advanced use' :)

TheTechnobear, 2015-07-29:
ok, its not in build.sh, I'll do that and commit it

closed 2015-07-29 by JohannesTaelman

JohannesTaelman, 2015-07-29:
Travis is working on a fresh build, up in around 5 minutes...

commit 2976494c (2015-07-29): Merge pull request #172 from TheTechnobear/master

FW:compile in build.sh, fixes #170
commit 3b019bd1 (2015-07-29): FW:compile in build.sh, fixes #170