axoloti/axoloti #171: Low Resolution App Icon

vormplus, 2015-07-29:
I've noticed that the app icon on OSX only contains a low resolution version.

@JohannesTaelman If you send me a vector graphics version of the Axoloti logo, I'll create the .icns file.

TheTechnobear, 2015-07-29:
it may be we need pngs as well (for windows/linux), I used what I had when i did the installers :) 
they are all stored in  - src/main/java/resources

vormplus, 2015-07-29:
I've sent the new .icns file and PNG files to Johannes.

You can probably use the same PNG files for Windows/Linux. These are the image sizes, let me know if you need other ones for the other platforms.

Filename | Image size
-------- | ----------
icon_16x16.png | 16 x 16 px
icon_16x16@2x.png | 32 x 32 px
icon_32x32.png | 32 x 32 px
icon_32x32@2x.png | 64 x 64 px
icon_128x128.png | 128 x 128 px
icon_128x128@2x.png | 256 x 256 px
icon_256x256.png | 256 x 256 px
icon_256x256@2x.png | 512 x 512 px
icon_512x512.png | 512 x 512 px |
icon_512x512@2x.png | 1024 x 1024 px

JohannesTaelman, 2015-07-29:
Thanks, committed splash and .icns in resources
not sure where to put other icons

TheTechnobear, 2015-07-29:
there are is an axoloti_icon.png in the same directory which is used...
not sure which size is the best for windows/linux, perhaps check them all in src/main/java/resources
and I will try with linux difference sizes :)

TheTechnobear, 2015-07-29:
actually, don't worry. whist I'm here , I might as well do it :)

TheTechnobear, 2015-07-29:
forget that, I though they were attachment to this issue, but they don't appear to be :)

TheTechnobear, 2015-07-29:
mac osx icon works - much nice :)

TheTechnobear, 2015-07-29:
ok linux I'm using 128x128 png, whats odd is the icon looks fine in the Ubuntu launcher but still 'crunchy' in the menu (sorry don't know whats its called :) ) ... but it doesn't matter even if I use the 512x512, so I suspect that is something to do with it converting it perhaps?  
anyway, no matter, if you lock to the launcher it looks good :)

vormplus, 2015-07-30:
I've just scaled the large icon down automatically for the smaller sizes. I'll redesign the smaller icons by hand so the shape of the logo is aligned on the pixel grid.

On Mac OS X, the OS chooses which size of the icon it displays. Might be possible on Linux as well to use the smaller icon for the menu and a larger one for the launcher.

closed 2015-08-01 by JohannesTaelman