axoloti/axoloti #179: bundle: file associations

JohannesTaelman, 2015-07-30:


TheTechnobear, 2015-07-30:
oops, been meaning to add that... added it now, and testing it

TheTechnobear, 2015-07-30:
I need to specialise the Application for this to work on the Mac, as it does not use the command line it uses com.apple.eawt.Application.setOpenFileHandler,
but this requires I also the build a little to ensure the AxolotiMacApp only is compiled and used on the Mac.... not difficult, but not tonight :)

TheTechnobear, 2015-07-31:
odd travis build on linux is throwing this error:
BUILD FAILED
/home/travis/build/axoloti/axoloti/build.xml:298: fx:info doesn't support the nested "association" element.

works perfectly fine on my Linux box (ubuntu 14.04) , perhaps an older version of JDK8?

I've commented it out in build.xml for now, and add javac -version to see what version JDK travis has
Im using : javac 1.8.0_45

TheTechnobear, 2015-07-31:
travis
$ javac -version
javac 1.8.0_31
perhaps try to get them to upgrade?  (doesnt seem that old, but its seems the only cause)

TheTechnobear, 2015-07-31:
ok, tried to update travis, the following is rrequired:
echo oracle-java8-installer shared/accepted-oracle-license-v1-1 select true | sudo /usr/bin/debconf-set-selections && sudo add-apt-repository ppa:webupd8team/java && sudo apt-get update -qq && sudo apt-get install oracle-java8-installer

there are a few hurdles:
a) there is only an deb package from ppa:webupd8team, so you need to add their repo - no issue 
b) unfortunately they throw a warning up, (i think during apt-update) which i cannot get around
c) then we need to auto accept the oracle licence, (which the first part of this statement should do) 

as i cannot get around (b) , I've rolled this back for the time being.