axoloti/axoloti #190: Midi: midi channel on top level patch

TheTechnobear, 2015-08-02:
Ive just stumbled across something which i think is pretty 'illogical'

if I have a top level patch , that contains a poly patch...
the top level patch only received channel 1 (or whats configured) but the poly patch can receive 1-16 

similarly I can create a mono patcher, and put midi channel attr on it, and it receives on channel 2.

I think it would be far more logical, that midi channels are 'filtered'  
(and it would be better if we selected a 'range' of channels not just one.

this would mean usually top level patches would be OMNI  (though you could filter if you wish),
and then subpatches would filter down as required.

this would also allow the top level patch to have midi objects which could filter based on channel.
(e.g. I could have a keyb channel object) 

the reason this is all useful, is when you have a single midi port (say usb device), its much more very useful to be able to send different data on different channels.

my case here, was I wanted to send 'trigger' notes on channel 16, and other polyphonic notes for the instrument on 1-15

currently to do this, I have to create a subpatch with  channel attribute ...  (even if patch.settings.channel =16 it wont work, as the top level patch filters it out)
even having used axoloti for a bit, I found this pretty surprising/counter intuitive.

JohannesTaelman, 2015-08-02:
The origin of have a midi channel assigned to a patch is for midi CC mapping to parameters. Mapped parameters are currently only receiving CC. But full mapping implies reflection. Changing a parameter should induce a midi CC transmit (hence #27), so it can be captured in a sequencer, or a midi controller with motor faders or led ring encoders can follow parameter changes from other origins.
When the patch channel would be set to omni, to what channel should a mapped parameter change be reflected?
But this is also an open problem with multiple midi ports. It could reflect on all ports, but that is not elegant.

TheTechnobear, 2015-08-02:
nope, midi reflection is not a good idea... (imho)
I do not have any device that receives a CC and then retransmits it... and would not want this..
e.g. lets say my synth uses CC 73 for cutoff, so I sent CC73 to change it ... it will NOT re-broadcast it!
sure, if i turn the cutoff know, it will send CC 73, but that is because it is the source of the event.

If a user wants to send a CC because a paramater has changed, this should be an explicit choice, and should be then targeted at where they want it to go.

(bare in mind, midi thru, is a completely separate function in most synths and if you want 'soft' thru it usually has to be explicitly enabled)