axoloti/axoloti #192: Obj:patch/load - mute during load?

TheTechnobear, 2015-08-03:
I get some serious 'clicks' when I use patch/load to load a new patch from SDCARD.

as we are unable to carry any voices across the load, I think we should be muting during load, e.g.

 - mute the audio output , 
 - stop the old 'patch'
 - load new patch and start
 - un mute audio output.

TheTechnobear, 2015-08-03:
its possibly this could be related to initialising of patches, but Im getting it with every load ... even for patches that don't usually have issue when going live.

JohannesTaelman, 2015-08-03:
Could you test uncommenting 
StopPatch();
line 178 in patch.c?
not sure why it got commmented...

TheTechnobear, 2015-08-03:
neither helped nor hindered :) 

ok, I think its actually perhaps the init() bug...  as I can get it from the UI too.
it seems to be, I can start the same patch multiple time, no clicks,
 
but the first time I load a different patch, I get a loud click, 
then I can load that patch, without clicks.

so its seems to be switching patches... something to do with loading  a different xpatch going up.

(its quite possible, this was the cause, when I said last night I was getting artefacts when going live, but didnt know when)

closed 2016-08-11 by axoloti