axoloti/axoloti #201: Windows install guide - (Rescue Mode) - more clarification for Zadig driver replacement needed

geslotencirkel, 2015-08-06:
I have re-checked how I did the Zadig driver replacement while upgrading an axoloti board with pre-1.0 firmware and found that the instructions need to be more detailed, but also that there might be an error in the instruction.

I will point to the line of the instructions (located here : http://community.axoloti.com/t/installing-on-windows/36) where we will need more clarification.

"Instead, go to Program Files (x86) /axoloti_runtime and run Zadig, and select install driver."

1. "zadig_2.1.2.exe" is located in "C:\Program Files (x86)\axoloti_runtime\platform_win" as per current installation instructions and zip package structure.
2. Initially, the Zadig window is blank - requiring the user to select "Options > List All Devices"
3. The axoloti board while in DFU mode in Windows will be listed as STM32 BOOTLOADER in the dropdown menu.
*** Here is where I am a bit confused at what we should write ****
4. The STM Driver that is installed for the STM32 BOOTLOADER needs to be switched [ I am not sure about this since I switched it to the wrong one(?) and was able to flash correctly
5. Rescue Mode instructions state that the driver needs to be replaced with  one that is provided in the zadig application [there are actually 3 to choose from! I assumed I simply install the one that is already selected - "WinUSB (v6.1.7600.16385)" , but there is also "libusbK" and libusb-win32 (v1.2.6.0)

I propose we clarify that you need to list all devices, and to select the STM32 BOOTLOADER device and switch it to the correct driver name (I am sure Johannes knows which one is correct, I won't attempt to try them all as it can possible cause some issue - but I was able to flash correctly with WinUSB driver). If the STM32 BOOTLOADER device is now there, then Axoloti was no put into DFU mode correctly.

TheTechnobear, 2015-08-06:
k, i'll try to update now...

TheTechnobear, 2015-08-06:
ok, updated on, http://community.axoloti.com/t/installing-on-windows/36 
I updated this according to the installation Ive just done on Windows 10, and works fine.
(fyi I use winusb)

geslotencirkel, 2015-08-06:
cool, looks good to me - although i made some typos myself in my original post above, in the last sentence. corrected it now

closed 2015-08-06 by JohannesTaelman