axoloti/axoloti #209: GUI: Menu structure

JohannesTaelman, 2015-08-08:
"Library", "recent", and "help-library" are only in the mainframe window menubar, not in the patch window menubar, they should be added to the patch window menubar too.

TheTechnobear, 2015-08-08:
wondered how long till that was noticed.... 
actually probably most of  the items that exist in the console, should exist in the patch (e.g. open recent patches/preferences) ... I find it pretty tedious having to find the console window to do the functions that are on it.

However, (and the reason Ive not done anything about it) ...  I think much of this will change when we need to start moving many of the functions to the patch window to support multiple boards. 
issue #65

axoloti, 2016-01-17:
partially addressed in 69578b45691b15031130535d740da20d9e029b6b
(still need to do help->library)

commit 4506dcbf (2016-01-17): GUI: help library menu in patcher, closes #209
closed 2016-01-17 by JohannesTaelman