axoloti/axoloti #223: typo in platform_linux/upload_fw_dfu.sh

pierstitus, 2015-08-12:
dfu upload script for linux doesn't work, there is /bin on two places, the second should be removed.

    cd "${platformdir}/bin"
    ./bin/dfu-util ...

TheTechnobear, 2015-08-12:
k, i'll fix now

TheTechnobear, 2015-08-12:
are you using the latest version (either from repo, the latest axoloti_runtime... ?

TheTechnobear, 2015-08-12:
my bad, it was only recently broken... fixing now

TheTechnobear, 2015-08-12:
fixed, will need need users to take a new axoloti_runtime version (linux only)

TheTechnobear, 2015-08-12:
@pierstitus just as an 'aside'  (as this needed fixing) , there is no real reason to use dfu update on an axoloti board, except in failure cases. (perhaps discovery board?)

pierstitus, 2015-08-12:
My beta version didn't connect, probably old firmware or I broke something when trying some things, so I had to use the rescue flash option.

closed 2015-08-12 by pierstitus