axoloti/axoloti #226: Using midi/in/keyb mod in a subpatch makes Axoloti crash.

vormplus, 2015-08-14:
Just lost some time with a nasty bug while using **midi/in/keyb mod** in a subpatch. When I upload the patch to the Axoloti, everything compiles. The Axoloti crashes and disconnects when I fire a MIDI note.

These are the errors I get in the console:

	Control transfer failed: -1
	Control transfer failed: -9
	Ping: WaitSync Timeout, disconnecting now

When I replace **midi/in/keyb mod** with **midi/in/keyb**, my polyphonic synth works fine.

TheTechnobear, 2015-08-14:
I suspect this may be related to an issue I had the other day, which @JohannesTaelman worked out was due to modulation sources in polyphonic subpatches. (keyb mod has them, keyb doesn't) 
so looks likely

(btw, if you change the subpatch to monophonic does it stop crashing?)

vormplus, 2015-08-16:
I can confirm that it only crashes when the subpatch mode is set to Polyphonic.
Setting the subpatch mode to Mono works fine.

closed 2015-09-30 by JohannesTaelman