axoloti/axoloti #239: UI: CPU load high when board connected.

TheTechnobear, 2015-08-27:
java process consumes about 0.6% cpu when UI is loaded but axoloti is not connected (in the UI... it is connected via USB) 
as soon as we hit connect, this jumps to 26% (and seems to utilise alot of one virtual core), taking a patch live then increases this to around 44%.

this seems pretty high, even given the IO required...
(my soundplane/eigenharp which have MUCH more IO, use considerably less) 

this is happening on both mac and linux not testing on windows.

TheTechnobear, 2015-08-27:
profiling shows its basically spending all its time in the receiver... we need to determine if its in a tight loop, or if we are sending/receiving too much data, or too frequently.
(an early thought, it perhaps update frequency of pingers/dials etc should be configurable... also supporting the 'remote' when its not open is also not required.)

JohannesTaelman, 2015-08-27:
Like in preferences->poll interval?
I get around 6% cpu usage on Windows 7, with a Core i7-920 cpu (getting old...)
The issue seems platform dependent, I suspect usb4java...
Not sending remote data is probably a good idea, but not a solution for this issue, the data rate is pretty low.

TheTechnobear, 2015-08-29:
poll interval preferences - yes ... though this was just an initial thought, not looked at the reality of it yet.

remote data - agreed, actually I think this is tied to another change, I want the updateUI call to be replaced with a ptr to func, so we can have a 'callback chain', a bit like signal(), then a remote object, would populate this as required. (linked in with 'controller objects') 

agreed, I also suspect its usb4java, hence why I said it may not be a quick fix.
(only thing i thought though, is it seems like a thin wrapper/JNI interface into libusb, and libusb performance is usually fine for me with other (non-java) products)
I'll do a test on win10 to see if I see the same issue, but thats not a conclusive test due to me using vmware... but will be enlightening.

Endres, 2016-04-17:
I can confirm, that using Linux (4.4.5-1-ARCH, libusb 1.0.20-1) and the current git state of axoloti, the CPU load is quite high (>1) whenever the axoloti is connected inside the GUI. I also suppose it could only be the part that connects java with libusb, because everything else should not be platform dependent, as this bug yet only shows up on Mac and Linux systems.

bafonso, 2016-06-30:
hitting 25-35% on my macbook pro, definitely crazy high and draining the battery fast. As if the java bug of forcing usage of the discrete GPU wasn't bad enough.. :-)

axoloti, 2016-08-19:
addressed in 0e10dbcaab880c991434885484404c21c933367c and 481100af49d089168bd69e0b99e5a5e3d5c372f3
untested on linux so far