axoloti/axoloti #246: Objects becomes invisible if you exceed Axoloti Patch Window limit.

jaffasplaffa, 2015-09-24:
Yesterday I noticed that if I drag objects far to the right side of Patch window the object becomes invisible. It is still there but you cannot see it. You can still mark all objects (Apple+a) and move it into the visible part of the patch window again. So it doesnt dissapear, only become invisible.

[image]

TheTechnobear, 2015-09-24:
drag sound be stopped at limit - width of component (i.e. 5000 - width)
alternatively we could extend the canvas as we reach the edge (possibly might flicker a lot due to redraws, so might have to do in 'chunks', say add another 500 pixels) , which may be useful for REALLY big patches.

jaffasplaffa, 2015-09-24:
It is not really that big of a deal, but I noticed it while working on a 16 part sequencer for Blofeld synth with loads of objects. Just thought I wanted to let the developer know :)