axoloti/axoloti #247: GUI: Better File Open/Save dialog window on Mac OS X

vormplus, 2015-10-02:
Left: Axoloti file dialog.
Right: Default Mac OS X File Dialog.

I'm mainly missing the left sidebar to browse my hard drive more quickly.

[image]

The current Axoloti file dialog also doesn't respond to keypresses. If I press the 'p' key for instance, the dialog should scroll to the first patch with a filename that starts with the letter p. Makes it a lot easier to locate a certain patch when you have lots of them in a directory.

vormplus, 2016-02-17:
This seems to be an issue with Swing on the Mac. Looks like AWT file dialogs work better.
https://developer.apple.com/library/mac/documentation/Java/Conceptual/Java14Development/07-NativePlatformIntegration/NativePlatformIntegration.html#//apple_ref/doc/uid/TP40001909-212646-TPXREF127

TheTechnobear, 2016-02-17:
unfortunately, this has always been the problem with java, its cross-platform, so it looks equally meh on all platforms ;)  honestly, I cant see me looking at this anytime soon...

I think Im much more likely to work on a dedicate patch/object browser which will be linked to the libraries, so try to avoid the OS file browsing entirely.

a much bigger issue, Ive been trying axoloti on my new(ish) MBP, thats running 10.11, and there are numerous issues display issues... e.g. dialogs popping up in the wrong place, wires not dragging correctly... I suspect its caused by the retina display.