axoloti/axoloti #248: Xorg fatal IO error 11

dromer, 2015-10-03:
When playing around in the patcher (for instance going through the 'Getting Started' book) the IDE completely crashes my Xorg.

Running on current debian sid and axoloti-1.0.3
Tried both:
```
$ java -version
java version "1.7.0_85"
OpenJDK Runtime Environment (IcedTea 2.6.1) (7u85-2.6.1-3)
OpenJDK Server VM (build 24.85-b03, mixed mode)
```
And:
```
$ java -version
openjdk version "1.8.0_66-internal"
OpenJDK Runtime Environment (build 1.8.0_66-internal-b01)
OpenJDK Server VM (build 25.66-b01, mixed mode)
```

This results in:
```
XIO:  fatal IO error 11 (Resource temporarily unavailable) on X server ":0"
after 7609 requests (7605 known processed) with 0 events remaining.
```

dromer, 2015-10-03:
Using the jre from oracle it seems much more stable:
```
$ java -version
java version "1.8.0_60"
Java(TM) SE Runtime Environment (build 1.8.0_60-b27)
Java HotSpot(TM) Server VM (build 25.60-b23, mixed mode)
```

Maybe we can mention this compatibility in the instructions.

dromer, 2015-10-20:
I still get this occasionally. Maybe it coincides with other graphical applications I am running, but I am uncertain of this.

This could very well be an unstable Xorg because current debian sid is going through a major gcc-migration. However I have only had this with the Axoloti Patcher.
It particularly happens in 'Live' mode when twiddling some knobs.

JohannesTaelman, 2015-10-22:
> when twiddling some knobs.
This may hint at the manipulations of the mouse pointer when adjusting a dial.
It is hiding the pointer, and changing its position, not very common in applications.

dromer, 2015-10-22:
As mentioned in issue #249 the cursor often jumps down tot he bottom of the screen when selecting a knob (which pulls down the value as well).
Maybe this is related?

dromer, 2015-10-22:
Hmm, today X actually crashed when I turned off live mode and was working in another program.

dromer, 2015-10-27:
Ok, I reverted my system to debian testing/stretch (should have slightly less breakage).

This still gave issues with openjdk7, but with openjdk8 it's smooth sailing so far.
Have been running, and playing with, the patcher for a couple days now without any issues.

(I think with openjdk7 I also had the mouse-cursor jumping away when turning knobs (to the left in this case), but with openjdk8 this is no issue)

closed 2016-08-19 by axoloti