axoloti/axoloti #249: Cursor becomes invisible

vormplus, 2015-10-10:
I sometimes have the issue where the cursor goes missing when a patch is Live, making it really hard to tweak dials.  The cursor is still there somewhere, since I see patch cords change color when I hover them.

Not sure how to reproduce this bug, only happens every once in a while.

Operating System: Mac OS X 10.9.5.

keymanpal, 2015-10-10:
Have to say I saw this happen more then once (late nights, me delusional?) 
Cant reproduce consistently… ;-(

Mac OS X 10.7.5

keymanpal, 2015-10-16:
This tend to happen more when you control a knob with shift+control...

dromer, 2015-10-20:
Hmm, I regularly have the cursor jump down the screen when I try to change a knob. It then also gets pulled down (and impossible to turn back up).

JohannesTaelman, 2015-11-19:
So this commit does not repeatedly hides the cursor when adjusting a dial, only once. 
@dromer do you use a multi-screen setup? Or a virtual desktop? When adjusting a dial, the mouse pointer is hidden and moved back constantly to the center of the dial. This avoids the need to re-grab the dial after adjusting.

dromer, 2015-11-19:
I do often have a multi-screen setup, but since I moved to debian Stretch I haven't noticed any more problems. Will report otherwise if they return.

dromer, 2015-12-19:
I can confirm that in my multi-monitor setup (linux, X11, fluxbox) the mouse-pointer shoots off the screen (some particular direction).
With single-monitor the patcher works fine on my laptop.

dromer, 2016-05-02:
With 1.0.9 this is still an issue (mouse cursor shooting off in multi-monitor when turning dials).